verilator/verilator #2067: Executable scripts missing shebang lines

zdave, 2019-12-23:
Various executable scripts (eg bin/verilator) are missing shebang lines (eg #!/usr/bin/env perl). AIUI some shells fall back to running an executable file as a shell script if exec fails, so the scripts do work if run with eg bash. They do not work in many other cases. eg Trying to run verilator from a Python script fails with "Exec format error" as Python does not have an equivalent fallback path:

```
$ python -c 'import subprocess; subprocess.call(["verilator"])'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
```

Replacing the magic at the top of the scripts with just "#!/usr/bin/env perl" works for me. I have no idea if this is an acceptable solution or not as it's not clear to me what the purpose of the magic is.

wsnyder, 2019-12-23:
Thanks for the report.

The magic is to handle operating systems where /usr/bin/env does not exist or a special version of perl is being used, so I am reluctant to change the shebang, as it'll take a lot of other experiments to get it stable again, if it's even possible.

This is pretty much only a Python problem common to using subprocess with any script.  When using subprocess, please just use "shell = True" when you call subprocess, and all should work.

Can you confirm that fixes your issue and is satisfactory?

zdave, 2019-12-23:
I actually noticed this issue because running verilator in the fish shell does not work. I just thought the Python example was an easy way to demonstrate the problem.

commit a2911ed9 (2020-01-10): Fix shebang breaking some shells. Closes #2067.
closed 2020-01-10 by wsnyder