verilator/verilator #2070: Strange width mismatch error

nanduraj1, 2019-12-27:
Strange width mismatch error.  LHS and RHS are both having same width. But each expects the other to be different!!! This is a deadlock

```
%Warning-WIDTH: ../../src/include/../../src/common/madaptor/sa_st_width_convert.sv:162: Operator SUB expects 32 or 7 bits on the LHS, but LHS's SEL generates 3 bits.
                                                                                      : ... In instance sa_st_width_convert_tb.dut
            ss_src_if.data[(CNTR_WIDTH'(RATIO-1)-CNTR_WIDTH'(cntr))*IN_WIDTH +: IN_WIDTH] <= data;
                                                ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: ../../src/include/../../src/common/madaptor/sa_st_width_convert.sv:162: Operator SUB expects 32 or 7 bits on the RHS, but RHS's VARREF 'cntr' generates 3 bits.
                                                                                      : ... In instance sa_st_width_convert_tb.dut
            ss_src_if.data[(CNTR_WIDTH'(RATIO-1)-CNTR_WIDTH'(cntr))*IN_WIDTH +: IN_WIDTH] <= data;
                                                ^
%Error: Exiting due to 2 warning(s)
```

May we assist you in trying to fix this yourself?
yes

wallento, 2019-12-29:
Thanks for your report. Can you please produce a minimal working example (plain verilog code) to reproduce the error? Thanks

nanduraj1, 2019-12-29:
The following code gives error 

### Code
```
module sa_st_width_convert
  (
   input i_clk
   );

  localparam IN_WIDTH =  512;
  localparam OUT_WIDTH = 64;

  logic [IN_WIDTH-1:0]      data;
  logic [OUT_WIDTH-1:0]     data_out, data_out2;

  always_ff @(posedge i_clk) begin
      data_out2 <= data[(9'(3) - 9'(1)) * 9'(OUT_WIDTH) +: OUT_WIDTH];  // This does not give error
      data_out <= data[(3'(3) - 3'(1)) * OUT_WIDTH +: OUT_WIDTH]; // This gives error
  end

endmodule
```

### Error
```
%Warning-WIDTH: sa_st_width_convert.sv:16: Operator SUB expects 32 or 7 bits on the LHS, but LHS's CONST '3'h3' generates 3 bits.
                                                                                                         : ... In instance sa_st_width_convert_tb.dut
      data_out <= data[(3'(3) - 3'(1)) * OUT_WIDTH +: OUT_WIDTH]; 
                              ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Warning-WIDTH: sa_st_width_convert.sv:16: Operator SUB expects 32 or 7 bits on the RHS, but RHS's CONST '3'h1' generates 3 bits.
                                                                                                         : ... In instance sa_st_width_convert_tb.dut
      data_out <= data[(3'(3) - 3'(1)) * OUT_WIDTH +: OUT_WIDTH]; 
                              ^
%Error: Exiting due to 2 warning(s)
```

### What I understood

in `(3'(3) - 3'(1)) * OUT_WIDTH `, It's actually the multiplier that has width mismatch between LHS and RHS

wallento, 2019-12-30:
Thanks, confirmed. I will look into it.

wallento, 2020-01-03:
@wsnyder I believe the problem is the following: https://github.com/verilator/verilator/blob/master/src/V3Width.cpp#L3672

In this situation we have a MUL with lhs SUB and rhs VARREF->VAR. The VAR is the parameter, sized at 32/min:7 (still need to understand why..). SUB is correct at 3/3, and it then fails because 3<7. The logic behind it seems a bit strange to me, but I seem to have a fundamental issue with understanding it: Min is supposed to be the minimum acceptable width and MUL gets this from the parameter and thus doesn't accept SUB's 3. I don't really understand why MUL cannot accept [2:0]*[6:0].

The math in this code works out, so I believe there is something wrong in the mininum calculation.

wallento, 2020-01-03:
Error reproduced here: https://github.com/verilator/verilator/tree/issue-2070

wallento, 2020-01-03:
The error may also be in the SUB path. It's the SUB of two CASTSIZE.

Using the CASTSIZE with the calculated result directly it works:

    assign slice = array[3'(2) * ARRAY_SLICE +: ARRAY_SLICE];

wsnyder, 2025-09-21:
The warning was because the multiplication with out width is 32 bits, so the warning while perhaps confusing is correct that a 32-bit number is desired on both left and right of the subtract.  Not expecting to do more on this so closing.