verilator/verilator #2073: Module i/o ports of unpacked array causing error in verilator + cocotb

nanduraj1, 2019-12-29:
### Verilog file
```
module arr_test ();

  logic din [0:15];

  array_test array_test_inst(.din(din));

endmodule

module array_test(
    input din [0:15]
);

endmodule
```
### Error

```
make[1]: Entering directory '/home/nndurj/Work/scratch/cocotb/examples/adder/tests'
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module arr_test --vpi --public-flat-rw --prefix Vtop -o arr_test -LDFLAGS "-L `dirname /home/nndurj/Work/scratch/cocotb/examples/adder/tests/build/libs/x86_64/libvpi.so` -lvpi -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/nndurj/Work/scratch/cocotb/examples/adder/tests/../hdl/adder.v /home/nndurj/miniconda3/envs/cocotb_verilator/lib/python3.7/site-packages/cocotb/share/lib/verilator/verilator.cpp
CPPFLAGS="-DVL_TIME_PRECISION_STR=1ps -std=c++11" make -C sim_build -f Vtop.mk
make[2]: Entering directory '/home/nndurj/Work/scratch/cocotb/examples/adder/tests/sim_build'
g++  -DVL_TIME_PRECISION_STR=1ps -std=c++11 -I.  -MMD -I/home/nndurj/softwares/verilator/include -I/home/nndurj/softwares/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o verilator.o /home/nndurj/miniconda3/envs/cocotb_verilator/lib/python3.7/site-packages/cocotb/share/lib/verilator/verilator.cpp
/usr/bin/perl /home/nndurj/softwares/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vtop.cpp > Vtop__ALLcls.cpp
g++  -DVL_TIME_PRECISION_STR=1ps -std=c++11 -I.  -MMD -I/home/nndurj/softwares/verilator/include -I/home/nndurj/softwares/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Vtop__ALLcls.o Vtop__ALLcls.cpp
In file included from Vtop__ALLcls.cpp:3:0:
Vtop.cpp: In static member function ‘static void Vtop::_settle__TOP__1(Vtop__Syms*)’:
Vtop.cpp:135:9: error: invalid array assignment
         = vlTOPp->arr_test__DOT____Vcellinp__array_test_inst__din;
         ^
Vtop.cpp:137:9: error: invalid array assignment
         = vlTOPp->arr_test__DOT____Vcellinp__array_test_inst__din;
         ^
/home/nndurj/softwares/verilator/include/verilated.mk:182: recipe for target 'Vtop__ALLcls.o' failed
make[2]: *** [Vtop__ALLcls.o] Error 1
make[2]: Leaving directory '/home/nndurj/Work/scratch/cocotb/examples/adder/tests/sim_build'
/home/nndurj/miniconda3/envs/cocotb_verilator/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:44: recipe for target 'sim_build/arr_test' failed
make[1]: *** [sim_build/arr_test] Error 2
make[1]: Leaving directory '/home/nndurj/Work/scratch/cocotb/examples/adder/tests'
/home/nndurj/miniconda3/envs/cocotb_verilator/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Error 2
```

May we assist you in trying to fix this yourself?
Yes

wallento, 2019-12-30:
Hi, sorry for being unresponsive by mail. I will check into it. Thanks for your patience

wallento, 2019-12-30:
What version of Verilator are you using? I ran it on the current HEAD and got no error. Can you please provide the output of

    verilator --version

wallento, 2019-12-30:
Here is the test I created to reproduce the problem: https://github.com/verilator/verilator/commit/182a4ab6dd4f88ac845c40b9b2aab41161b20b73

nanduraj1, 2019-12-30:
Hi Stefan, 

I'm using latest verilator and cocotb, both checkout out from github and installed.
verilator version is `Verilator 4.024 2019-12-08 rev v4.024-1-g6f52208`

if you replace the adder.v example in cocotb with the following and run the sim using verilator, you should get this error.

```
module arr_test ();

  logic din [0:15];

  array_test array_test_inst(.din(din));

endmodule

module array_test(
    input din [0:15]
);

endmodule
```

wallento, 2019-12-30:
Strange, let me have a look at it, thanks for reporting

wallento, 2019-12-30:
Okay, confirmed. The problem comes with `--public-flat-rw`. I will look into it.

wallento, 2019-12-30:
@wsnyder the problem is this line: https://github.com/verilator/verilator/blob/master/src/V3Slice.cpp#L130

What happens here is that there is an AssignAlias and a normal AssignW. The latter is properly sliced to 16 assignments, while the AssignAlias is not. Do you remember why this line is there? Sorry, I cannot grasp what it does. Thanks for your help.

wallento, 2019-12-30:
When I remove this line I get:

```
%Error: test.v:11: Internal: Unexpected Call
    input din [0:15]
          ^~~
%Error: Internal Error: test.v:11: ../V3AstNodes.h:2642: Unexpected Call
    input din [0:15]
          ^~~
```

Need to dig deeper of course, but I believe it would be very helpful if you could provide your thoughts about what may go on here..

wsnyder, 2019-12-31:
If an alias was sliced it wouldn't be an alias (another name for same signal) any more so it checks that.

How this relates to the bug I don't know, if you could continue to look that would be much appreciated.  (V3Slice was originally contributed code.)

wallento, 2019-12-31:
I see. So what seems to happen:

 - It shouldn't generate an alias when both signals are public, as they will be separate later
 - It seems that it is correctly emitted, but has skipped the slicing

Seems easy to fix by solving the first issue. Probably make sure the latter doesn't happen in other circumstances too. I will get onto it in the next days.

By the way, have you considered using memcpy where slicing code is emitted at the moment?

wsnyder, 2019-12-31:
Memcpy is likely slower as the compiler will not unroll/optimize as well as it will need to spill the registers. Also would be confusing.

Anyhow I'm working on class support and realizing that medoum term slicing perhaps will just be a special stayement that copies appropriate more complicated types.

wallento, 2019-12-31:
Tracking the issue in branch: https://github.com/verilator/verilator/tree/issue-2073
The test reproduces the error, it may take a couple of days to get to it.

closed 2020-01-03 by wallento

nndurj, 2020-01-03:
Thank @wallento

commit 50fcf2dc (2020-01-03): No assign alias for unpacked public variables

Public variables are all emitted in the C code and unpacked arrays
arrays are sliced up for this. After inlining public unpacked array
assignments should not be alias assignments but actual assignments, so
that they are sliced and hence emitted properly.

Fixes #2073
PR #2089 (2020-01-03): No assign alias for unpacked public variables
Public variables are all emitted in the C code and unpacked arrays
arrays are sliced up for this. After inlining public unpacked array
assignments should not be alias assignments but actual assignments, so
that they are sliced and hence emitted properly.

Fixes #2073
commit 924fe235 (2020-01-03): No assign alias for unpacked public variables (#2089)

Public variables are all emitted in the C code and unpacked arrays
arrays are sliced up for this. After inlining public unpacked array
assignments should not be alias assignments but actual assignments, so
that they are sliced and hence emitted properly.

Fixes #2073