verilator/verilator #2074: Python Support

BracketMaster, 2019-12-29:
The README suggests that verilator has Python support. I also found this #1363. It seems that CMAKE support was merged but Python support was not?

This would be extremely useful. I have been working on the Python Migen and nMigen HDLs for some time now rolling my own PyVerilator solution. 

Besides supporting just HDL, I advocate that Python is just a natural language to write unit tests and perform debugging in - owing to its fantastic REPL. Lots of verilator use cases include interfacing with matrices which python has fantastic support for via numpy. Also, I've seen some Machine Learning hardware accelerators in development. The industry DeFacto Machine Learning libraries(namely Google's Tensorflow and Facebook's PyTorch) are written in Python. So I think that it is very logical to include Python support for verilator.

BracketMaster, 2019-12-29:
OK. I just found here that it failed some tests. I didn't see that before.
https://github.com/verilator/verilator/pull/3

patstew, 2020-01-06:
The relevant issue is here: #1489 and the PR is at: #3
I need to get round to fixing the last few issues to get it into mainline at some point.

wsnyder, 2020-01-09:
Thanks for your report, as noted we agree this would be useful & think #1489 and #3 can track this.

closed 2020-01-09 by wsnyder