verilator/verilator #2081: Internal: Deprecation of features

wallento, 2019-12-30:
Hi,

I am wondering if you are interested to implement some kind of deprecation management. The idea I have is very basic and roughly involves:

 - Define a lifecycle of a deprecation, where there is a grace period (tbd) where it gives an info, after that a warning and then a feature is removed
 - Add a warning type "DEPRECATION" that can be ignored selectively using what we already have
 - Document deprecation of a feature with a schedule in the release notes
 - Add infrastructure for it and a dist test that automatically checks if there is anything scheduled to be deprecated, then it can be extended or ultimately removed.

Not sure if that is a major issue, but there are a few deprecated things in the code base. They don't hurt, but it seems like a nice thing and good starter task maybe.

wsnyder, 2019-12-30:
As a first step/experiment I think I would agree/suggest we try adding DEPRECATED as a warning (instead of the info) for the msg stuff you just pulled.

sc_clock is probably anothet candidate.

Will think about the other good suggestions.

wsnyder, 2020-01-20:
About to commit adding DEPRECATIONS section to docs/verilator, and noted items I think are eligible.

commit 1acff950 (2020-01-20): Commentary: Add deprecations section to docs. Closes #2081.
closed 2020-01-20 by wsnyder