verilator/verilator #2082: Upgrading from 4.016 to 4.024 introduces a verilation failure

oharboe, 2019-12-30:
There appears to be a new limit introduced after 4.016 and possibly as late as 4.024.

Is this intentional?

This is generated Verilog where the ridiculously wide signal is ultimately narrowed down in synthesis optimization, so the physical signal is not as wide as the below.

```
Step #6: %Error: DataCache.v:9212: Unsupported: Width of number exceeds implementation limit: 8191'd0
Step #6:   assign _GEN_7533 = {{8191'd0}, state_slave_0_writeData};  
```

wallento, 2019-12-31:
This is probably it: https://github.com/verilator/verilator/commit/9eaec3b5c10e6c3973d1e10b8a46cc795dbcbca3

There is a limit now, the number seems quite arbitrary.

wallento, 2019-12-31:
Can you please try from branch https://github.com/verilator/verilator/tree/issue-2082

oharboe, 2019-12-31:
@wallento This error actually pointed out a problem in my code, which I fixed. A problem that synthesis hid again, so not really observable, but I didn't intend to use 8000 bits wide integers :-) 

I believe what's needed here is a test-case in the Verilator test-suite, not manual testing.

wallento, 2019-12-31:
The easiest solution is to increase the `MAX_WIDTH` to a number that is properly handled by `atoi`: https://github.com/verilator/verilator/blob/master/src/V3Number.cpp#L35

@wsnyder is the implementation limit arbitrary or may it make sense to increase it to some actual max? Large numbers will wrap as the comment says, and that's a case needed to catch, but I am wondering what the MAX is supposed to catch.

mikeyangsiv, 2020-01-19:
We're also hitting this in Chisel/FIRRTL generated Verilog code.  It generates literals like `16384'd1`.

oharboe, 2020-01-19:
@mikeyangsiv  same for me: Chisel/FIRRTL generated Verilog code. I didn't have that wide integers really, I was relying on synthesis to eliminate unused bits. For me it made sense to rewrite the Chisel code not to generate so wide integers.

wallento, 2020-01-20:
Can you please check the `issue-2082` branch?

mikeyangsiv, 2020-01-21:
@wallento Building and running one of the generated designs that was previously failing with `issue-2082` now passes 🎆 . Thanks!

wallento, 2020-01-21:
Thanks for confirming. As a note to future generations: Use `--max-num-width` to increase the limit on the size.

closed 2020-01-21 by wallento

oharboe, 2020-01-21:
Thanks! I like things being solved with more speed than haste. This seems to put it to bed properly.

commit 22088c90 (2020-01-21): Set maximum number width (#2128)

Adjust the maximum number width to 64K. Add --max-num-width option to
adjust this setting.

Closes #2082