verilator/verilator #2084: loFIRRTL frontend

wallento, 2019-12-31:
In my love for parsers I have started playing around with a Verilator frontend for FIRRTL a while ago. It allows to directly feed chisel generated *.fir files into verilator without going through Verilog. The major advantage is that warnings and error messages can carry the scala fileline information at no extra cost. Constructing the AST from firrtl is pretty straightforward.

Do you think this would be a useful addition?

wsnyder, 2019-12-31:
For the FIRRTL nieve (e.g. me) can you point to some example files?

omasanori, 2020-01-25:
I am not a FIRRTL expert, but the `samples/` directory of [Treadle](https://github.com/freechipsproject/treadle) may be useful. Unfortunately, the [spec](https://github.com/freechipsproject/firrtl/blob/master/spec/spec.pdf) lacks a precise definition of LoFIRRTL. Perhaps it does not matter since full FIRRTL is not so high-level compared to Verilog HDL, I guess?

davidmlw, 2020-04-02:
Nice, @wallento where is your play ground? Will it has benifit to accept hiFIRRTL?

wsnyder, 2021-02-21:
Closing as unlikely there will be work on this in the medium term, long term maybe!

closed 2021-02-21 by wsnyder