verilator/verilator #2086: Bad Parameter resolution in generate block

nndurj, 2020-01-01:
### Failing Code 

Here `if` condition is inside `always_comb`

```systemverilog
module t();

    localparam D_WIDTH = 16;
    logic [D_WIDTH-1:0] a, b, c;

    generate
        always_comb begin
            if (D_WIDTH > 2*D_WIDTH) begin
                {a, b} = c[2*D_WIDTH-1:0];
            end
        end
    endgenerate

endmodule
```

### Error
```
verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module t --vpi --public-flat-rw --prefix Vtop -o t -LDFLAGS "-L `dirname /home/nndurj/Dev/OpenSource/cocotb_test/tests/build/libs/x86_64/libvpi.so` -lvpi -lgpi -lcocotb -lgpilog -lcocotbutils"  /home/nndurj/Dev/OpenSource/cocotb_test/tests/../hdl/test.v /home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/share/lib/verilator/verilator.cpp
%Error: /home/nndurj/Dev/OpenSource/cocotb_test/tests/../hdl/test.v:9: Extracting 32 bits from only 16 bit number
                                                                     : ... In instance t
                {a, b} = c[2*D_WIDTH-1:0];
                          ^
%Warning-SELRANGE: /home/nndurj/Dev/OpenSource/cocotb_test/tests/../hdl/test.v:9: Selection index out of range: 31:0 outside 15:0
                                                                                : ... In instance t
                {a, b} = c[2*D_WIDTH-1:0];
                          ^
                   ... Use "/* verilator lint_off SELRANGE */" and lint_on around source to disable this message.
%Error: Exiting due to 1 error(s), 1 warning(s)

```

### Passing Code

Here `if` condition is outside of `always_comb`

```systemverilog
module t();

    localparam D_WIDTH = 16;
    logic [D_WIDTH-1:0] a, b, c;

    generate
        if (D_WIDTH > 2*D_WIDTH) begin
            always_comb begin
                {a, b} = c[2*D_WIDTH-1:0];
            end
        end
    endgenerate

endmodule
```

wallento, 2020-01-03:
Thanks for reporting, it looks indeed inconsistent, but I am not sure if this is easy to fix. Also, two out of three commercial simulators behave exactly the same.

nndurj, 2020-01-03:
Well, I guess this is not a big deal.

wallento, 2020-01-03:
It is certainly a nice-to-have and it matches exactly the kind of "but the simulator should be able to.." cases I also stumble over. On the other hand it seems not trivial to fix at a first glance..

wsnyder, 2020-01-09:
I believe verilator is correct as-is.  As written the code basically becomes

            if (0) begin
                a = c[0 + 16];
                b = c[31 : 16];
            end

And Verilator chooses to warn even for always-off blocks as the code still is "wrong". Note technically accessing out of bounds is IEEE legal so the one commercial simulator that takes this probably just isn't warning as IMO it should.

closed 2020-01-09 by wsnyder