verilator/verilator #2091: --lint-only does too much work

Magpie7, 2020-01-08:
I'm compiling a large design with --lint-only, it consumes ~150GB of RAM while doing so.
If I add -O0 to the command line it uses ~5GB.

I suggest that --lint-only could turn of optimisation, without the user having to think about it?

wallento, 2020-01-08:
Yes, this sounds right. I will look into it.

Magpie7, 2020-01-08:
A little more information.
--lint-only and -O0 gives ~5G mem usage, but hadn't completed after two hours.
--lint-only uses 150GB of mem, but finishes in 30 minutes.

Verilator v4.014

wsnyder, 2020-01-08:
For the run that finishes, please run with --stats and attach the __stats.txt file, thanks.

Magpie7, 2020-01-08:
Here you are.

[blah__stats.txt](https://github.com/verilator/verilator/files/4037516/blah__stats.txt)

wsnyder, 2020-01-08:
Please try "-Ox" without -O0, as most of the memory is from the Expand optimization.

Magpie7, 2020-01-08:
-O1 without --lint_only has completed verilation!  Thanks.
-O1 and --lint_only crashed.  I'll try and reproduce and run with --debug/--stats tomorrow.

Magpie7, 2020-01-09:
A more clear summary of what I see.

--lint-only
	Maximum resident set size (kbytes): 147122032
	Elapsed (wall clock) time (h:mm:ss or m:ss): 26:59.87

--lint-only -O0
	Doesn't finish in 4 hours

--lint-only -O1
    Maximum resident set size (kbytes): 147123560
	Elapsed (wall clock) time (h:mm:ss or m:ss): 29:02.42

--lint-only -Ox
	Maximum resident set size (kbytes): 42924912
	Elapsed (wall clock) time (h:mm:ss or m:ss): 20:34.77

-Ox (i.e. actually generate C code)
	Maximum resident set size (kbytes): 43882252
	Elapsed (wall clock) time (h:mm:ss or m:ss): 23:52.22

wsnyder, 2020-01-09:
Pushed disabling -Ox.

A 4-1 expansion due to -Ox is expected, FWIW. Thanks for the data.

commit aac02c1e (2020-01-09): Fix expand optimization slowing --lint-only. Closes #2091.
closed 2020-01-09 by wsnyder