verilator/verilator #2093: $fwrite formats and $time issues

agrobman, 2020-01-09:
Hi,

I see some problems with $fwrite formats and $time:

my verilog code looks like:
 
```
                    $fwrite (el, "%10t : %6s 0 %h %h %s\n",$time(), $sformatf("#%0d",commit_count), 
                           trace_rv_i_address_ip[31+i*32 -:32], trace_rv_i_insn_ip[31+i*32-:32],
                           wb_dest[i] !=0 ?  $sformatf("r%0d=%h", wb_dest[i], wb_data[i]) : "");
```
irun and vcs prints it this way:

```
       235 :     #1 0 80000000 b0201073 
       245 :     #2 0 80000004 b8201073 
       255 :     #3 0 80000008 ee0000b7 r1=ee000000
```

but verilator ( I use 4.020 version):

```
         0 : #1 0 80000000 b0201073 
         0 : #2 0 80000004 b8201073 
         0 : #3 0 80000008 ee0000b7 r1=ee000000
```
i.e. $time - first field returns always 0,
verilator doesn't honor %6s format (#N field) - prints only visible chars in the string.

Any ideas?

wsnyder, 2020-01-10:
As to sizes of %s, that wasn't supported, I just fixed this in git, please try it.

As to time being zero, I can't explain that, are you sure that time isn't really zero?  That is maybe some # delays are in the other simulator and ignored by Verilator?  The other possibility is you have a time divisor set.

agrobman, 2020-01-10:
I'm getting VCD waves and time in ns is shown ( with simvision) 

"
The other possibility is you have a time divisor set.
"
 can you explain this, please? what's the time divisor?

wsnyder, 2020-01-10:
I was referring to defining VL_TIME_MULTIPLIER, but looking at the code that wouldn't make it print zero.

agrobman, 2020-01-10:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

Can you check on your side that $time yields  not zero results? Or give working  example?

wsnyder, 2020-01-10:
the test_regress/t/t_display.pl test among many others test $time.

agrobman, 2020-01-10:
How do  I run this?

]% /wdc/apps/veripool/verilator/4.020/verilator-4.020/test_regress/t/t_display_time.pl

======================================================================

dist/t_display_time: ==================================================

-Skip: dist/t_display_time: scenario 'dist' not enabled for test

dist/t_display_time: %Skip: Skip: scenario 'dist' not enabled for test

%Error: No such file or directory obj_dist/t_display_time/Vt_display_time.status, at ./driver.pl line 720.

Compilation failed in require at ./driver.pl line 698.

%Warning: dist/t_display_time: driver.pl _read_status file missing: obj_dist/t_display_time/Vt_display_time.status

dist/t_display_time: FAILED: driver.pl _read_status file missing: obj_dist/t_display_time/Vt_display_time.status

%Error: Permission denied writing obj_dist/driver_20200109_171442.log, at ./driver.pl line 372.

wsnyder, 2020-01-10:
/wdc/apps/veripool/verilator/4.020/verilator-4.020/test_regress/t/t_display_time.pl --vlt

--vlt means run on Verilator.  Alternatively --nc or --vcs etc.

More info on testing in docs/internals.adoc

agrobman, 2020-01-10:
No luck :
% /wdc/apps/veripool/verilator/4.020/verilator-4.020/test_regress/t/t_display_time.pl --vlt
======================================================================
vlt/t_display_time: ==================================================
%Error: No such file or directory obj_vlt/t_display_time/Vt_display_time__main.cpp, at ./driver.pl line 1462.
Compilation failed in require at ./driver.pl line 698.
%Warning: vlt/t_display_time: driver.pl _read_status file missing: obj_vlt/t_display_time/Vt_display_time.status
vlt/t_display_time: FAILED: driver.pl _read_status file missing: obj_vlt/t_display_time/Vt_display_time.status
%Error: Permission denied writing obj_dist/driver_20200109_172223.log, at ./driver.pl line 372.

wsnyder, 2020-01-10:
The permission denied suggests you are running it frome some global install you cannot write to.  Try a git clone of the repo under your home dir, then run the test from there.

agrobman, 2020-01-10:
Hi,

I have some trouble to install the tool locally – seems some libraries are not installed on our systems. Also I’m not a programmer and not familiar with Linux SW creation/installation.
Is there any hack possible to define work directory outside installation area?

wsnyder, 2020-01-10:
In theory you should be able to run the test in a personal checkout of Verilator, but point at a different install for the binary, so you don't need to build Verilator in your personal checkout.

    export VERILATOR_BIN=`which verilator`
    test_regress/t/t_display_time.pl

agrobman, 2020-01-11:
Hi,

could please review our RISCV swerv (on github) demo TB setup ?: it generates exec.log file where first field  is time and this one is always zero.

the clock and reset for verilator version are generated with tb cpp file in the testbench directory,
run commands are in tools/Makefile under verilator-run target

wsnyder, 2020-01-11:
Your testbench never changes time to non-zero.

    testbench/test_tb_top.cpp:
    vluint64_t main_time = 0;
    double sc_time_stamp () {
       return main_time;
    }

wsnyder, 2020-01-11:
With the format issue fixed and $time resolved I think Verilator issues here are closed.

closed 2020-01-11 by wsnyder

agrobman, 2020-01-11:
How the time supposed to be changed? With any clock edge?

agrobman, 2020-01-11:
Is there way to use vcs /verilog TB with internal clock generators, using  # delays as is, without this tb cpp file?

Are $dumpfile/$dumpvars/$finish supported?

wsnyder, 2020-01-11:
>How the time supposed to be changed? With any clock edge?

That's up to your testbench.  Given I think you have just one clock incrementing each edge probably makes sense.

>$dumpfile/$dumpvars/$finish supported?

No, no, yes.  Please see documentation and just try stuff.

agrobman, 2020-01-11:
Thanks a lot for all your support!