verilator/verilator #2094: SV dynamic types support

agrobman, 2020-01-10:
Are there any plans to support SV dynamic types?

support of SV queues and associative arrays would be very appreciated. ( at least of basic types and structures of basic types ).

Need something to model sparse memories for CPU verification.

wsnyder, 2020-01-10:
SV queues and assoc arrays are both now supported, perhaps you have an old version.

wsnyder, 2020-01-10:
P.S. Note it isn't in a release version, but git master.

closed 2020-01-10 by wsnyder

agrobman, 2020-01-10:
> CAUTION: This email originated from outside of Western Digital. Do not click on links or open attachments unless you recognize the sender and know that the content is safe.

What version does support these ? I have 4.020

agrobman, 2020-01-10:
I’m getting this with 4.020

%Error: /wdc/proj/riscv/mustang/users/alexander.grobman/eh1.1/swerv_eh1.1/testbench/ahb_sif.sv:46: syntax error, unexpected ']', expecting "'{"
int kuku[int];
                     ^

wsnyder, 2020-01-10:
You need to check it out from git, not use a release.  Or wait a few days-ish for the next release.

agrobman, 2020-01-10:
I’m not sure I can use master version as we provide opensource solution for RISCV community.

agrobman, 2020-01-10:
Could you provide any timeline for next release?

wsnyder, 2020-01-10:
Will try to get it out before Monday.

wsnyder, 2020-01-11:
released today.

wsnyder, 2020-01-11:
released today.

agrobman, 2020-01-11:
Any documentation what is supported now and how?
We immediately need associative arrays of bytes, indexed by unsigned integer with a possibility of load from a hex file with $readmemh.
Are these features supported now?
Queues are 2nd priority.

wsnyder, 2020-01-11:
Didn't implement readmem yet, please file a feature bug.  Please see how it does otherwise.

agrobman, 2020-01-11:
We need associative arrays to implement sparse processor memory, where code and data sit in random locations. (we don’t want allocate 4GB static RAM) We really don’t have  time to write Verilog parser for hex files – our due day for new cores release is Jan 21. Is it possible to get this functionality before Jan 21?

Anyway, how to know what is implemented/supported from dynamic types?

Is there plan to support classes?

BTW, is there windows version of verilator?

wsnyder, 2020-01-11:
> We need associative arrays to implement sparse processor memory, where code and data sit in random locations. (we don’t want allocate 4GB static RAM) We really don’t have  time to write Verilog [...]

We aren't paid for our work, so if you need something by a date you need to do it yourself. However I appreciate the Swerv project and your management, so will look at $readmem support - you need to file a feature bug to request this.

>Anyway, how to know what is implemented/supported from dynamic types?

The best thing to do is try to use a feature (e.g. associative arrays) and if it doesn't do what you want file a bug.

>Is there plan to support classes?

First support is in a branch, but it will be a few months until usable, and likely a year before everything related to classes works.

>is there windows version of verilator?

See the documentation and #2078

agrobman, 2020-01-11:
I know you work for free. There is no pressure, I just wanted to set expectations what we can do until our release ..

Regarding dynamic types, at least give a list of what is supported :

Dynamic arrays,
Queues,
Associative arrays, etc.

Are all arrays/queues functions supported ?
What ordinary types can be used in queues/arrays?

Are there any release notes?

wsnyder, 2020-01-11:
See ChangeLog in the root directory.  bin/verilator documents some things but isn't exhaustive, best is to try stuff yourself.

wsnyder, 2020-01-11:
See **Changes** in the root directory. bin/verilator documents some things but isn't exhaustive, best is to try stuff yourself.