verilator/verilator #2095: Strange error report

agrobman, 2020-01-10:
I had strange error report, where I had a dotted reference to a internal signal of an instance.
The tool complained that couldn't find this signal, while the problem was that the module definition file was missing ...

Only when I removed the "dotted" references the tool reported the module definition was missing!

wsnyder, 2020-01-10:
If you can attach an example we'll take a look.

This is all more complicated than it might seem, as a dotted reference might be inside a generate if to a module that is OK if it doesn't exist (because the generate if will prove to be false).

agrobman, 2020-01-10:
I have something like this:
file tb.sv:

module tb;

always @ ( posedge kuku.end)
 $display("kuku.finish=%h",kuku.finish);

kuku kuku(...);
endmodule
 
file kuku.sv

module kuku( ...);
wire finish;
endmodule

wsnyder, 2020-01-10:
Sorry, I need an example which runs through the tool and shows the issue.

agrobman, 2020-01-10:
File tb.v :

module tb;
initial begin
    $display("a=", dut.a);
end

dut dut();
endmodule

% verilator --cc tb.v
%Error: tb.v:3: Can't find definition of 'a' in dotted variable: 'dut.a'
    $display("a=", dut.a);
                       ^
%Error: Verilator internal fault, sorry.  Consider trying --debug --gdbbt
%Error: Command Failed /wdc/apps/veripool/verilator/4.020/bin/verilator_bin --cc tb.v

Other simulators will complain first that dut isn’t defined …

agrobman, 2020-01-11:
It also core dump!

wallento, 2020-01-11:
This was fixed between 4.020 and 4.022. Please update to a newer version and confirm its fixed for you.

wsnyder, 2020-01-12:
I'm pretty sure this is still non-optimal, I get:

    %Error: /svaha/wsnyder/d/a.v:3: Can't find definition of 'a' in dotted variable: 'dut.a'
    $display("a=", dut.a);
                       ^

Instead of it complaining about not finding dut.

wallento, 2020-01-12:
Ah, right, I see the issue now

wsnyder, 2020-12-13:
Fixed finally!  It gives double error but that seems an ok tradeoff in this case.

```
%Error: t/t_inst_missing_dot_bad.v:9:22: Dotted reference to cell that refers to missing module: 'missing'
    9 |       $display("a=", missing.a);
      |                      ^~~~~~~
%Error: t/t_inst_missing_dot_bad.v:9:30: Can't find definition of 'a' in dotted variable: 'missing.a'
    9 |       $display("a=", missing.a);
      |                              ^

```

closed 2020-12-13 by wsnyder

agrobman, 2020-12-13:
why is the "cell"? Verilog calls this object as 'instance' .. It still requires some time to comprehend what's the problem!
instead of "missing module" better to say "Dotted reference to not defined object 'missing', I think.
In future it could be missing interface definition or class or any other complex object like structure ..

wsnyder, 2020-12-13:
It was cell only because no one ever pointed that out ;)  (Verilator's calling it "cell" predates IEEE ;)  Fixed dozens of places.

I also now say module/interface; it can't be anything else due to the way the error is detected.