verilator/verilator #2097: --protect-lib fails with clang

wsnyder, 2020-01-11:
./configure CXX=clang++

Then testing fails the protect tests due to:

     ccache clang++  -I. -Wall -Wextra -Wfloat-conversion -Wthread-safety -Werror -MMD -I/verilator/include -I/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -fPIC   -c -o secret.o secret.cpp
        secret.cpp:18:32: error: implicit conversion from 'long' to 'int' changes value from 2256956716 to -2038010580 [-Werror,-Wconstant-conversion]
        int expected_hash__V = 2256956716;

(Perhaps int should be unsigned?) Commenting that out gives a bigger issue:

    ccache clang++  -I. -Wall -Wextra -Wfloat-conversion -Wthread-safety -Werror -MMD -I/verilator/include -I/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -fPIC   -shared -o libsecret.so secret.cpp Vt_prot_lib_secret__ALL.a verilated.o secret.cpp Vt_prot_lib_secret.h /verilator/include/verilated_heavy.h /verilator/include/verilated.h /verilator/include/verilatedos.h /verilator/include/verilated_dpi.h /verilator/include/verilated_sym_props.h /verilator/include/vltstd/svdpi.h
    clang: error: treating 'c-header' input as 'c++-header' when in C++ mode, this behavior is deprecated [-Werror,-Wdeprecated]
    clang: error: cannot specify -o when generating multiple output files
    Vt_prot_lib_secret.mk:61: recipe for target 'libsecret.so' failed

The .h files shouldn't be on the compiler line.

Thanks

toddstrader, 2020-01-11:
I can sort of reproduce the first error by running t_prot_lib after configuring with clang++-8.  However, I get the message as a warning and not an error.  I'm guessing this is a function of what configure finds on my system.  I'm also guessing this is why Travis hasn't found this before.  In any case this fixes the issue and works with gcc too:
https://github.com/toddstrader/verilator-dev/tree/prot-lib-clang

For the second part, I'm not sure how to reproduce that.  I don't see any header files being used (except with -DVM_PREFIX_INCLUDE=) when running t_prot_lib (if that is in fact what I should be looking at).

Please let me know if that patch fixes your issue.  If so, I can push and if not I can try to reproduce if you can help me figure out what to try next.

Thanks.

wsnyder, 2020-01-11:
Huh, with that I don't see the other problem. Very odd. Anyhow, I'd suggest "U" to match other places we do this, then feel free to push upstream. (If you wish by a pull request as it seems your master is behind.)

Note I have another pull request to turn on warnings as errors in travis, will merge when passes.

toddstrader, 2020-01-11:
Done.

closed 2020-01-11 by toddstrader