verilator/verilator #2103: FST Parameter Dumping Stopped Working (in a certain situation?)

RichardMyersGV, 2020-01-12:
Hi there,

I'm experiencing a strange problem.  Recently, parameters dumped via --trace-params all show up as XXX in GTKWave when dumping in the FST format. Real parameters show up as nan.

Parameter values get used properly in the Verilog code (assignments, widths, etc.).  The issue doesn't affect VCD dumping.  I've determined that it's not something that was introduced to Verilator recently (I've tried from 4.026 back to 4.014).  It affecs all parameters regardless of where they are declared in the code.  

If I include my header file, which contains all of the parameters, in a simple project, the parameters all show up correctly in the FST file.  This isn't a real apples to apples test though, as it's compiled under Linux to a Linux executable.  In my real application, the Verilated output is compiled by Ming32 (32-bit) into a Windows DLL and run under Wine.  I could probably make a better apples to apples test, but it's a non-trivial task, and I'm hoping that someone familiar with Verilator internals might be able to point me in the right direction.

I've Verilated with '--debug --stats-vars' and can't see anything obviously amiss.  The project Verilates and compiles cleanly, with no warnings.

Everything otherwise simulates and displays in the FST file as expected.  So whatever the issue is, it just seems to affect parameter dumping in FST (maybe in a Ming32-generated DLL and maybe under Wine).  It worked fine until recently. 

Any idea how I might determine the cause of this issue?  I supposed it could've started with a Wine update (DLL is run under Wine), or even a recent Ming32 update.

Thanks,
Richard

wsnyder, 2020-01-12:
Suggest first convert the FST to VCD with fst2vcd and see if the VCD displays OK. If so GTKwave bug.

Else, add print statements to verilated_fst.h to the fullBit/Bus/Double etc (whatever data type it is) and see if Verilator passes the right data.  If so, GTKwave bug else verilator bug and we'll proceed from there.

RichardMyersGV, 2020-01-13:
Hi,

Tried that.  It looks the same when converted to VCD using fst2vcd (parameters show up as XXX or nan in GTKWave).

As far as adding the print statements, the issue affects any and all data types as long as they are declared as 'parameter' or 'localparam'.  

In looking at  verilated_fst_c.h, ... there are no functions related to parameters, and I'm not clear about how to proceed, or what I'm looking for.  How would I know whether an output of the function call is a real wire or a real parameter?   
void fullDouble(vluint32_t code, const double newval) {
  // printf ("Something...");
  chgDouble(code, newval); }
Will it print every time a double changes during simulation?  Just once for a particular parameter?  How will I know it's a parameter verses a real wire?

Also, I guess there's the further complication of it being a DLL under Windows.  I think I can dump to a debug file...  But I'm still unsure of what I'm looking for.

I wish I was more adept at Verilator internals.  Well, I'll try some things and hopefully learn something ...

Richard

wsnyder, 2020-01-13:
fullDouble is called when first dumping data in a new file.  chgDouble when the value changes (which it shouldn't for a parameter).  The code arg indicates what signal; in declDouble print the code and name, then you know when fullDouble is called with a code what the name will correspond to.

wsnyder, 2020-02-24:
Any update on this issue, or can we close it?

RichardMyersGV, 2020-02-24:
Sorry, still haven't got to digging more deeply into this.  I'm totally 
buried at the moment.  It's still an annoying issue in 4.028.  If I were 
up on Verilator internals and debugging, I'm sure I could find the issue 
in no time...

There was one point recently, where it healed and parameters showed up 
correctly in the FST file.   I'm sure it's getting confused about 
something in my header file...  I'll try to make some time for this in 
the near future.

There was also a simple build error with 4.028 under Mingw32.  I had to 
delete #define __STDC_FORMAT_MACROS from verilatedos.h, line 225, as it 
was already defined at verilatedos.h, line 278.  I haven't got around to 
sending a patch ... sorry.

Thanks,
Richard

RichardMyersGV, 2020-02-25:
Hi,
declDouble, fullDouble (and Bits, Bus, etc.) all look OK, with the expected values, and yet still show up as XXX or nan in the FST file.
declDouble(): code = 4106, name = $unit VIN_PER_LSB
declBus(): code = 3426, name = $unit ADC0_WD, dtypenum = -1, fstVarType = 2
declBus(): code = 3546, name = top dt fc FM_PREC, dtypenum = -1, fstVarType = 2
fullDouble(): code = 4106, newval = 13.4409
fullBus(): code = 3426, newval = 10, bits = 32
fullBus(): code = 3546, newval = 16, bits = 32

[image]

Any more suggestions on where to look?  I'm Using GTKWave 3.3.104 and Verilator 4.028.

Thanks,
Richard

RichardMyersGV, 2020-02-25:
The issue also affects wires, like:
output wire [31:0] rtl_version
assign rtl_version = 32'h02_23_20_00;
and in an upper module...
assign DLL_Version = rtl_version;
[image]
Strange...
Richard

RichardMyersGV, 2020-02-25:
In the previous message, please disregard the difference in DLL_Version and rtl_version.  I updated the version and didn't reload the waveforms...
Richard

wsnyder, 2020-02-25:
Given the calls to verilated_fst are correct and the fst2vcd output is wrong this is very likely a bug in GTKwave's fst writer code or related libraries.  You'll probably need to either make a test case for Tony Bybell, and/or debug the library yourself and feed Tony the resulting fixes.  Contact him if you need tips as I haven't poked in there much.

wsnyder, 2020-02-25:
CC @gtkwave

gtkwave, 2020-02-25:
Parameters should look just the same as dumping of non-parameters.  It's just different enum for the value/string associated with the variable similar to a reg declaration.

It sounds to me like you're seeing a problem with constant assigns that never change in value.

When is Verilator emitting these?  If it's immediately after variable creation time, there might be an issue.  In fact, I can almost guarantee Wilson's dumper code is doing something along those lines as I can't think of any other way for this kind of initialization breakage outside of params not being written in the first place.  :-)

Usually all the variables/scopes/attributes are created, then you emit time and variables later on without ever emitting new variables.  Think of $enddeffinitions/$end in VCD.  The way I wrote the writer, I allowed for late signal adds, but in 11 years, nobody has actually tried it...

Probably what needs to be done is in fstDestroyMmaps()
lseek(__fd, cur_offs, SEEK_SET);
needs to be SEEK_END (or completely omitted), not SEEK_SET in order to accommodate late signal adds.  I don't know 100% if that will work.  I'll have to experiment here and write some writer interface testcode.

It's going to be something along those lines where the time 0 value table is getting obliterated.

-Tony

gtkwave, 2020-02-25:
That lseek sounds wrong too when I look at what's being passed to it...I'll have to play with it.

-Tony

gtkwave, 2020-02-25:
I wrote my own testcase to verify my suspicions.  If this is what Verilator is doing, it's why it doesn't work.  I'll see if there is an easy fix in the writer.

-Tony

int main(int argc, char **argv)
{
void *ctx = fstWriterCreate("zzzzz.fst", 1);
fstHandle r[8];

r[0] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r0", 0);
r[1] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r1", 0);
r[2] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r2", 0);
r[3] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r3", 0);

fstWriterEmitValueChange(ctx, r[0], "0000");
fstWriterEmitValueChange(ctx, r[1], "0001");
fstWriterEmitValueChange(ctx, r[2], "0010");
fstWriterEmitValueChange(ctx, r[3], "0011");

r[4] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r4", 0);
r[5] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r5", 0);
r[6] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r6", 0);
r[7] = fstWriterCreateVar(ctx, FST_VT_VCD_REG, FST_VD_IMPLICIT, 4, "r7", 0);

fstWriterEmitTimeChange(ctx, 0);

fstWriterEmitValueChange(ctx, r[4], "0100");
fstWriterEmitValueChange(ctx, r[5], "0101");
fstWriterEmitValueChange(ctx, r[6], "0110");
fstWriterEmitValueChange(ctx, r[7], "0111");

fstWriterEmitTimeChange(ctx, 100);

fstWriterEmitValueChange(ctx, r[0], "zzzz");

fstWriterEmitTimeChange(ctx, 200);

fstWriterEmitValueChange(ctx, r[1], "zzzz");

fstWriterEmitTimeChange(ctx, 300);

fstWriterClose(ctx);
exit(0);
}

wsnyder, 2020-02-25:
Verilator should always declare before it puts out a change, then once there is a change never declare anything again. This is because this is what VCD requires.

I don't know if there's an assertion that protects against misuse of this however, @RichardMyersGV is there only a single design which you call trace once on (versus multiple models you are tracing together)?  Can you run this on Linux?  My first suspicion would be a windows problem versus something general.

RichardMyersGV, 2020-02-25:
Hi,
Yes, just one design.  I've added the header file with all the 'defines and parameters to a Verilator example project under Linux, and all is as expected (correct).  

With some work, I could make a Linux project that makes an executable from the whole project.  It won't do anything useful, but would be enough to make an FST file.

I suspect you're right about it being a Windows issue.  Funny thing is that it works sometimes, and worked for a long time before recently.  It could've started after a MingW update, or Wine update, or RTL update.  I just don't know.  Dang it.

Thanks,
Richard

RichardMyersGV, 2020-02-26:
It does the same thing under real Windows (vs Wine).  It's a DLL built with MingW in either case.  Running Verilator with --trace-fst or --trace-fst-thread doesn't change the issue.  Verilator is always run under Linux, in case anyone's wondering.

gtkwave, 2020-02-26:
I don't really care about the code generation by Verilator under Linux if that is the case.  I care about the compiled "verilated code" that is the executable.  It's possible a Windows code path in the FST writer is blown in one of the mingw #ifdefs.  I need to see actual examples that fail that I can reproduce them on my end.  Otherwise, I can't fix this.  It's possible there's something as weird as a missed buffer flush condition because Windows doesn't have the same shared memory IPC, so I emulate it.

I'll keep digging into the X problem for initial values as I thought that worked (and could be a side issue separate of this).

Thanks,
-Tony

wsnyder, 2020-02-26:
@gtkwave Just a guess & not sure if it's used in windows but I noted fstapi.c calls write() but doesn't check the return value.  Note write() is NOT required to write all the bytes that you request so should always check the return value and move buffer appropriately.

Tony, perhaps put a printf on the write() return value and see if it matches the amount requested.

wsnyder, 2020-02-26:
Think similar bug with read()

wsnyder, 2020-02-26:
Meant to say *Richard*, please try a printf on the write() return value and see if it matches the amount requested.

wsnyder, 2020-02-26:
Very unrelated, but @gtkwave I just noticed in fstapi.c there's a bunch of variables beginning with __ used.  Ideally these should be renamed as double underscores are reserved for the implementation, so this is illegal according to ANSI, and if you run the google code checker it'll probably warn (or at least some tool does, as I got dinged on this before).

gtkwave, 2020-02-26:
It's likely going to be something weird in the __MINGW32__ ifdef'd routines, specifically with tempfile generation.  

For example, I had to do some workaround in the past:

#ifdef __MINGW32__
        setvbuf(fcomp, (char *)NULL, _IONBF, 0);   /* keeps gzip from acting weird in tandem with fopen */

...it would not shock me if there are more areas for Windows that require such handling.

-Tony

RichardMyersGV, 2020-02-26:
Here's a surprise ... I converted the project to build a purely Linux executable.  It makes an FST file.  The FST file has the same issue as when I build it under Windows.  The parameters under $unit are all XXXX.  And parameters within modules are XXXX

RichardMyersGV, 2020-02-26:
Didn't get any incomplete read/writes in fstRead/fstWrite (compared read/write return vals with cnt...).

gtkwave, 2020-02-26:
How about the generated source for the code that makes all the fstWriter calls or a minimal chunk of verilog + driver code that reproduces the error and can be built/run on my end? I have no idea what is or isn't occurring based on the description. I use VCS for my day job.

In the least, the same behavior for Linux cuts out all the Windows-specific code, which is good.

-Tony

RichardMyersGV, 2020-02-26:
Unfortunately, I have not yet been able to create a test case that shows the issue...

For what its worth, if I define FST_DEBUG, all I see is:
[rmm@crank verilator] $ ./dll_main
Enabling waves into top.fst...
**FSTAPI  | value chains: 166**
[rmm@crank verilator] $ GTKWAVE | Reloading waveform...
FSTLOAD | Processing 3564 facs.
FSTLOAD | Built 527 signals and 3037 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
GTKWAVE | ...waveform reloaded

wsnyder, 2020-02-26:
Might be a fair bit of work, but perhaps put a printf() at each call into the fst library in verilated_vcd.h/cpp that printf's the C++ code that call represents. Then if you take all of that output, wrap it in a main(), and compile it you should be able to see the issue. Anonymize and post.

RichardMyersGV, 2020-04-05:
Darn, I just spent many hours trying to debug this and got nowhere.  It's killing me.  Such a PITA not having anything that is set at T=0 show up in the FST file.  Even something like:
wire one = 1'b1; 
shows up in the FST file as x.

For one parameter:

fstWriterCreateVar() (c2e190) name = BRN_STATES, len = 32, handle = 519
fstWrite() Wrote 1 items of 10 size
fstWrite() Wrote 1 items of 1 size
fstWrite() Wrote 1 items of 2 size
...
fstWriterEmitValueChange() (c2e190) val = 00000000000000000000000000000011, handle = 519
fstWriterEmitValueChange(FST_LIKELY)
fstWriterEmitValueChange(FST_LIKELY(len)) len = 32, xc->is_initial_time = 1

and that's it.  I never see a FstWrite or FstWriter function call.  And all subsequent fstWriterEmitValueChange() calls have xc->is_initial_time = 0.

I've tried making a subset of the project, with the same header file, and it always works fine.

Any idea how I can catch the issue?  I see Wilson's suggestion above, but I don't understand it, and it seems like there could be a more straightforward way that I can do myself.  

The above fprintf output is from a 300+MB file that is the result of just one top->eval() call.

Thanks,
Richard

wsnyder, 2020-04-05:
Perhaps you're calling trace() by accident before the first eval call?  I think if you called trace then eval then incremented time this might confuse things.

RichardMyersGV, 2020-04-05:
Thanks, this got me started in a new direction.  

Good grief!  I didn't call trace before an eval(), but I did have two tfp->dump(0) calls.  An init() function had an eval() and dump() at T=0, and the first main routine call was also at T=0.  Turns out you can call dump() multiple times with the same time, but can only have one tfp->dump(0).  The second call overwrites the first somehow (is_initial_time == 0, I guess).  

It would be nice to add a static flag to the VerilatedFst class, and/or something to dump(), that checks for this and returns immediately if the flag is set.  That, at least, would keep dummies like me from tripping over this situation.

I'm soooo excited to have my parameters and unchanging wires back again!!!

wsnyder, 2020-04-05:
Glad you found it!  Perhaps you could make a pull request since you have a model to test it, that asserts if two calls at any time stamp?

gtkwave, 2020-04-05:
I'm glad to hear the fix was found as well. It's good to see some sense of balance returned to the universe. :-)

-Tony

wsnyder, 2020-05-02:
Fixed earlier as noted above.

closed 2020-05-02 by wsnyder