verilator/verilator #2105: Error while accessing memory element through cocotb

lavanyajagan, 2020-01-13:
The following setup is being used:
```
cocotb-config: 1.3.0rc2
Verilator 4.025 devel rev v4.024-70-gf23fe8fd
Python 3.7
```
The example in the attached results in the following error. To reproduce: 
``$ cd adder_arr/tests; make SIM=verilator``
```
Traceback (most recent call last):
File "/usr/local/lib/python3.7/dist-packages/cocotb/log.py", line 198, in _log_from_c
  logger.handle(record)
File "/usr/lib/python3.7/logging/__init__.py", line 1524, in handle
  self.callHandlers(record)
File "/usr/lib/python3.7/logging/__init__.py", line 1586, in callHandlers
  hdlr.handle(record)
File "/usr/lib/python3.7/logging/__init__.py", line 894, in handle
  self.emit(record)
File "/usr/lib/python3.7/logging/__init__.py", line 1025, in emit
  msg = self.format(record)
File "/usr/lib/python3.7/logging/__init__.py", line 869, in format
  return fmt.format(record)
File "/usr/local/lib/python3.7/dist-packages/cocotb/log.py", line 174, in format
  return self._format(level, record, msg, coloured=True)
File "/usr/local/lib/python3.7/dist-packages/cocotb/log.py", line 111, in _format
  time_ns = get_sim_time('ns')
RecursionError: maximum recursion depth exceeded
Fatal Python error: Cannot recover from stack overflow.
```
[adder_arr.tar.gz](https://github.com/verilator/verilator/files/4053572/adder_arr.tar.gz)

The same test works with the ``cocotb-vcs`` setup

nanduraj1, 2020-01-13:
This error happens when we access unpacked array through cocotb. Another observation is that, if the array is packed and multidimensional, it does not even show up cocotb @wallento Is this related to #2073

wsnyder, 2020-01-18:
Suggest filing this against cocotb, as don't know if this will get seen here.

wallento, 2020-01-18:
Hi, there are a couple of missing things in VPI and/or public, that can take a while to add. @nanduraj1  can you please create a minimal Verilog example that doesn't work with cocotb? I understand this is an issue with unpacked arrays. I suspect they are not properly handled by `--public-flat-rw`, and the design ends up empty.

nanduraj1, 2020-01-18:
Hi @wallento, @lavanyajagan  has already attached a failing test. here is it again 

### Failing Test
`cd adder_arr/tests; make SIM=verilator`
https://github.com/verilator/verilator/files/4053572/adder_arr.tar.gz

wallento, 2020-01-18:
thanks. Interesting, I think it is supposed to work actually, I will look into it.

elgorwi, 2020-06-03:
@lavanyajagan ,
Please close this issue, fixes have just been merged.

wsnyder, 2020-06-03:
It's ok to close assuming we believe it fixed.  Feel free to reopen if there is still a problem.

closed 2020-06-03 by wsnyder