verilator/verilator #2106: Cannot find file containing module in search dir when module name >127 characters

AlexLee256, 2020-01-13:
As shown by the following test case, Verilator 4.016 cannot find the corresponding file for a module whose name is longer than 127 characters when given the search directory with -y:

`test.v`:
```
module test ();
 really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_long_module_name inst_e ();
endmodule
```
`really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_long_module_name.v`:
```
module really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_really_long_module_name ();
endmodule
```

Result:
> $ verilator --cc -y . test
> %Error: test.v:2: Cannot find file containing module: really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0
>         ... Looked in:
>              really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0
>              really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0.v
>              really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0.sv
>              obj_dir/really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0
>              obj_dir/really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0.v
>              obj_dir/really_really_really_really_real__Vhsha0o4sqIbRR6PPgOXJ23vHqAM4r0.sv
> %Error: Exiting due to 1 error(s)
> %Error: Command Failed /opt/verilator-4.016/bin/verilator_bin --cc -y . test

I suspect this could be related to the fix for [Issue 937](https://github.com/verilator/verilator/issues/937). 

The difference here is that the module with the absurdly long name is not defined in the same file in which it is used.

wsnyder, 2020-01-13:
The fix needed for this is non-trivial. My inclination would be to improve the error message to mention > 127 is not supported for automatic lookup, and leave improving this for later.  The workaround would be to simply put the filenames on the command line/-f file.  Do you see directly supporting finding these files as critical for some application?

AlexLee256, 2020-01-14:
I don't think it is critical to support directly finding the files as we can use your suggested workaround.

If we need our build scripts to retain the ability to search for files we can can use vhier to generate a file list from the search directories as the first step, and then feed the filenames to Verilator (I already checked that vhier does not suffer from the same limitation).

The use case arises because we have a tool which generates specialized versions of modules by "flattening" the parameters into the module name.

wsnyder, 2020-01-14:
I added cleaner error message.  Note the reason Verilator hashes like this to 127 characters is to leave room for prefix and a hash of parameter values, so your script might be advised to likewise to hash down to e.g. 100 or so.

closed 2020-01-14 by wsnyder