verilator/verilator #2108: VPI: Couldn't find root handle

nanduraj1, 2020-01-13:
### Problem
Some issues when I instantiate an interface. No error occures if I remove the instantiation

### HDL
```systemverilog
interface Bus;
  logic [7:0] addr, data;
  logic valid;

  modport src (output addr, output data, output valid);
  modport snk (input addr, input data, input valid);
endinterface

module t();

    Bus b1(); // No errors if I comment out this line

endmodule

```

### Error
```
        sim_build/t 
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:78   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:104  in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:340  in embed_sim_init                  Running on Verilator version 4.027 devel
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.0 from /home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1578942459
     0.00ns ERROR    cocotb.gpi                                  VpiImpl.cpp:447  in get_root_handle                 VPI: Couldn't find root handle t
     0.00ns ERROR    cocotb.gpi                                  VpiImpl.cpp:453  in get_root_handle                 VPI: Toplevel instances: t != __PVT__t__DOT__b1...
     0.00ns ERROR    cocotb.gpi                                GpiCommon.cpp:262  in gpi_get_root_handle             No root handle found
Traceback (most recent call last):
  File "/home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/__init__.py", line 175, in _initialise_testbench
    regression_manager.initialise()
  File "/home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/regression.py", line 128, in initialise
    self._root_name)
AttributeError: Can not find Root Handle (t)
     0.00ns ERROR    cocotb.gpi                                  gpi_embed.c:425  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:198  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
^C/home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/share/makefiles/simulators/Makefile.verilator:59: recipe for target 'results.xml' failed
make[1]: *** [results.xml] Interrupt
/home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/share/makefiles/Makefile.sim:79: recipe for target 'sim' failed
make: *** [sim] Interrupt

```

Possibly related to #1511, #2105

closed 2020-01-14 by nanduraj1

nanduraj1, 2020-01-14:
reinstalling cocotb and verilator solved this

Saikiran196, 2024-02-06:
I am getting this type of error only like COuldn't find the root handle

wsnyder, 2024-02-06:
@Saikiran196 I'd suggest posting an example to the CocoTB list instead.  If you find a problem related to Verilator, we'll need a self-contained example that does not use CocoTB - and please file a new bug.