verilator/verilator #2109: Unable to access interface instance through cocotb vpi

nanduraj1, 2020-01-14:
### Issue

Can't access interface instance through latest verilator+cocotb vpi. It was possible earlier on https://github.com/antmicro/cocotb-verilator-build

### HDL
```systemverilog
interface Bus;
    logic [7:0] a;
    logic b [7:0];

    modport src (output a, input b);
    modport snk (input a, output b);
endinterface

module t(
    input i_clk
);

    Bus b1();

endmodule
```

### Cocotb testbench
```python
import cocotb
from cocotb.triggers import Timer

@cocotb.test(skip=False)
def basic_test(dut):
    v = dut.b1.a
    yield Timer(10)
```

### Error
```
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:341  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:138  in _initialise_testbench           Running tests with cocotb v1.3.0 from /home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages
     0.00ns INFO     cocotb                                      __init__.py:155  in _initialise_testbench           Seeding Python random module with 1578975413
     0.00ns INFO     cocotb.regression                         regression.py:190  in initialise                      Found test test.basic_test
     0.00ns INFO     cocotb.regression                         regression.py:348  in execute                         Running test 1/1: basic_test
     0.00ns INFO     ..tb.test.basic_test.0x7fa8e8369890       decorators.py:253  in _advance                        Starting test: "basic_test"
                                                                                                                     Description: None
     0.00ns ERROR    cocotb.regression                         regression.py:327  in handle_result                   Test Failed: basic_test (result was AttributeError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/home/nndurj/Dev/OpenSource/cocotb_test/sv_interface_test/tests/test.py", line 6, in basic_test
                                                                                                                         v = dut.b1.a
                                                                                                                       File "/home/nndurj/miniconda3/envs/cocotb/lib/python3.7/site-packages/cocotb/handle.py", line 285, in __getattr__
                                                                                                                         raise AttributeError("%s contains no object named %s" % (self._name, name))
                                                                                                                     AttributeError: t contains no object named b1
     0.00ns ERROR    cocotb.regression                         regression.py:210  in tear_down                       Failed 1 out of 1 tests (0 skipped)
     0.00ns INFO     cocotb.regression                         regression.py:400  in _log_test_summary               *************************************************************************
                                                                                                                     ** TEST             PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     *************************************************************************
                                                                                                                     ** test.basic_test    FAIL            0.00          0.00         6.70  **
                                                                                                                     *************************************************************************

```
### Test
[sv_interface_test.tar.gz](https://github.com/verilator/verilator/files/4057298/sv_interface_test.tar.gz)

wsnyder, 2020-01-15:
Suggest filing this against cocotb, as don't know if this will get seen here.

wallento, 2020-01-18:
It is a Verilator issue I think as interfaces are not yet fully supported by the VPI runtime. Same is for structs which currently resolve to vpiReg and their elements are not accessible.

Those will be a bit heavy and take some time.

nanduraj1, 2020-01-18:
@wallento , this same test works fine with https://github.com/antmicro/cocotb-verilator-build, and we can read all the signals of the interface.

wallento, 2020-01-20:
I see that the signal got inlined. I will check why and what is missing on the Verilator side compared to the PoC

wallento, 2020-01-20:
Hi, can you please check branch `issue-2109`? I think it solves the problem.

nanduraj1, 2020-01-21:
Hi @wallento , this fix seems to solve the problem.

closed 2020-01-21 by wallento

WernerFS, 2024-02-11:
Hi, I reproduced the example using ```Verilator 5.020 2024-01-01 rev v5.020``` and ```cocotb 1.8.1```. It seems the problem with exposing interfaces is still there. I am unable to see ```b1``` under the ```dut```:

```
import cocotb
from cocotb.triggers import Timer

@cocotb.test(skip=False)
def basic_test(dut):
    print(dir(dut))
```

-----------------------------------------------------------------------------------------------------------

```
['_HierarchyObject__get_sub_handle_by_name', '__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_child_path', '_compat_mapping', '_def_file', '_def_name', '_discover_all', '_discovered', '_fullname', '_handle', '_id', '_invalid_sub_handles', '_len', '_log', '_name', '_path', '_sub_handle_key', '_sub_handles', '_type', 'get_definition_file', 'get_definition_name', 'i_clk']
```

Is there something else that needs to be done to expose interfaces to cocotb?

BTW here is my Makefile, not sure if it matters. 

```
SIM ?= verilator
TOPLEVEL_LANG ?= verilog
VERILOG_SOURCES += $(PWD)/top.sv

EXTRA_ARGS += --trace

all:
	rm -rf sim_build
	$(MAKE) sim MODULE=test TOPLEVEL=t

include $(shell cocotb-config --makefiles)/Makefile.sim
```

commit 3ee40407 (2020-01-20): Fix vpi scope naming

Dedot and shorten the name so that public modules and interface cells
are accessible too via VPI.

Fixes #2109
commit 08d88430 (2020-01-21): Fix vpi scope naming

Dedot and shorten the name so that public modules and interface cells
are accessible too via VPI.

Fixes #2109
commit 750d3ac1 (2020-01-21): Fix vpi scope naming

Dedot and shorten the name so that public modules and interface cells
are accessible too via VPI.

Fixes #2109
PR #2130 (2020-01-21): Fix vpi scope naming
Dedot and shorten the name so that public modules and interface cells
are accessible too via VPI.

Fixes #2109