verilator/verilator #2110: Include parameters and propagated values in XML

pieter3d, 2020-01-14:
Would be great for the xml elaboration output to contain parameter data. So the modules would contain elements for the declared parameters and (maybe) defaults, and the instances would contain elements for the overridden parameters with the overriding values.

I'm willing to help add this feature, though some pointers would be much appreciated.

wsnyder, 2020-01-14:
There are some potential complexities and limitations we can get into, but they could be overcome.  Can you describe the use case for this?

pieter3d, 2020-01-14:
The idea is to use Verilator's parser as a front end to a source browser/navigator. Seeing the actual values of parameters within a design would be very useful.

wsnyder, 2020-01-14:
Committed keeping the parameters, which is the easy part.  I'll close assuming this is enough, if not feel free to reopen and contribute a patch.

closed 2020-01-14 by wsnyder

pieter3d, 2020-01-15:
Thanks, this should help. Looks like it doesn't add the overridden values in the instance. I'll start digging to see how to add that.