verilator/verilator #2113: Setting Verilated::gotFinish() to true without $finish

norandomtechie, 2020-01-15:
I am writing a program that when Verilated with a given Verilog file, exchanges inputs and outputs with a script using stdin/stdout.  The Verilog file simply describes some form of hardware (simple stuff like shift registers and counters) and has no testbench code that will call $finish.  In my main.cpp file where I instantiate the top module using Vtop* top = new Vtop, how would I cause the simulation loop to correctly break with Verilated::gotFinish() when certain conditions are met (i.e. invalid input is received from the script, 5 minutes have passed without activity etc.)?

wsnyder, 2020-01-15:
Yes, calling gotFinished is a fine approach given this usage.  Alternatively if you want cross-simulator compatibility, call vpi_control(vpiFinish);

norandomtechie, 2020-01-15:
In my particular use case, I am not using VPI/DPI functions.
I tried compiling with gotFinished, and it gave an error: gotFinished is not a member of Verilated.  I found code in include/verilated.cpp that implied that you could also exit the loop by passing 'true' to the gotFinish call, i.e. Verilated::gotFinish(true).  I haven't tried it yet, but it definitely compiles.

wsnyder, 2020-01-15:
Yes, use Verilated::gotFinish(true), or call vl_finish(...), or call vpi_control(vpiFinish).

closed 2020-01-15 by wsnyder