verilator/verilator #2114: verilator generates C++ that doesn't compile: error: use of undeclared identifier 'VL_SHIFTR_WWQ'

solomatnikov, 2020-01-16:
Error:
```
VTestDriver.cpp:184747:5: error: use of undeclared identifier 'VL_SHIFTR_WWQ'
    VL_SHIFTR_WWQ(512,512,33, __Vtemp2768, __Vtemp2767, 
```

Generated C++:
```
    VL_SHIFTR_WWQ(512,512,33, __Vtemp2768, __Vtemp2767, 
		  (VL_ULL(0x1ffffffff) & ((QData)((IData)(
							  ((IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_v0_idx) 
							   << 7U))) 
					  + ((QData)((IData)(
							     (0x3ffffU 
							      & (((7U 
								   & (((IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_store)
								        ? (IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_src_idx)
								        : (IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_dst_idx)) 
								      & (((IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT___T_2081) 
									  >> 1U) 
									 | (1U 
									    & ((IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT___T_2081) 
									       >> 3U))))) 
								  << 
								  (7U 
								   & ((IData)(4U) 
								      - (IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_vtype_vsew)))) 
								 + (IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT__mem_i_beat))))) 
					     << (IData)(vlTOPp->TestDriver__DOT__testHarness__DOT__system__DOT__E27__DOT__vpu__DOT___T_2097)))));
```

wsnyder, 2020-01-16:
You probably have a >> by a 33 bit variable, which is a bit strange given one never can shift that much, but anyhow certainly legal.

Perhaps you could attempt to create a pull request that modifies one of the existing tests to show the issue, then add this function to verilated.cpp?

solomatnikov, 2020-01-16:
Yes, limiting the size of shift variable seems to solve this issue

closed 2020-02-24 by wsnyder