verilator/verilator #2115: Linux version  dependency

agrobman, 2020-01-16:
Is this true that
"
The latest version of Verilator requires a newer version of OS (RHEL 7) 
"
?

Our IT said so,  but we have only one server with it, and they plan to migrate to RHEL7 at the end of the year.

Is it possible to build verilator for older linux version?

we have something like this:
uname -a
Linux uls-ep-vhpcetx08 2.6.32-754.10.1.el6.x86_64 #1 SMP Thu Nov 29 15:44:01 EST 2018 x86_64 x86_64 x86_64 GNU/Linux

wsnyder, 2020-01-16:
The latest version has not (intentionally) changed the requirements.

It was tested to work on RHEL 6.4, except for --threads which has always needed C++11.

closed 2020-01-16 by wsnyder

agrobman, 2020-01-23:
Hi Wilson,

Our 4.027 build produces following:

verilator-4.027/bin/verilator_bin: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by verilator-4.027/bin/verilator_bin)

Any workaround for this?

wsnyder, 2020-01-23:
That's not something due to Verilator directly.  You probably built with a system with one glibc and are running on a different one, or didn't completely clean the build areas between versions.