verilator/verilator #2116: Why do we need `ifdef  VERILATOR in swerv EH1 RTL?

agrobman, 2020-01-17:
Hi,

Can you check please why do we need this `ifdef statements in 
design/pic_ctrl.sv file of the swerv DB?

Now I'm getting similar (?) error for our new cores
in memories preloading code :
"
%Error: /wdc/proj/riscv/mustang/users/alexander.grobman/ehx2/testbench/tb_top.sv:694: Can't find definition of 'dccm' in dotted signal: 'rvtop.mem.Gen_dccm_enable.dccm.mem_bank__BRA__0__KET__.dccm.dccm_bank.ram_core'
                                                                                    : ... In instance tb_top
    rvtop.mem.Gen_dccm_enable.dccm.mem_bank[0].dccm.dccm_bank.ram_core[indx] = data;
                                                              ^~~~~~~~
        ... Known scopes under 'ram_core': <no cells found>
        ... Known scopes under 'ram_core': <no cells found>

"
2nd dccm is named block of  implicit "generate" construct

like 
 (if PARM == 5) begin : dccm
            ram_5 dccm_bank ( ...);
 end
ram_core is verilog memory, modelling the RAM block.

Of course, this design runs with irun and vcs.

BTW, speaking of verilator TB, do we still need to sample finished output of verilog TB to stop simulation, or $finish will also work?

wsnyder, 2020-01-18:
Verilator doesn't support at present runtime-variable indexing into arrayed submodules.

The newer example you attach might be a different issue as it has a constant into ram_bank, and I presume ram_core is a variable.  Perhaps Verilator names the generates incorrectly, this area of the spec isn't well specified.  Compare the Known Scopes versus what you expect.

$finish has been supported for a long time.

agrobman, 2020-01-18:
Hi Wilson,

Thanks again for your help.

Regarding pic module `ifdef – I don’t think this was simulation time indexing of the instances – this construct is synthesizable, irun and vcs also do not support runtime variables to index arrayed   instances.

For the newer problem:

ram_core is Verilog memory array – variable;
what is “the Known Scopes”?

Interesting that very similar code for existing public swerv EH1 code does work in Verilator. The difference in RTL is that for new cores there is one more hierarchy of generate statements  with named block “dccm”.

We had to have it to make our RTL fully parameterizable to allow multiple instances of the CPU with different internal memories sizes.

Regards
Alex

wsnyder, 2020-01-18:
By "known scopes" I mean the error message tells you where it wasn't able to find something and what the next scopes it knows about are, generally this suggests what is not being found or is named strangely.

closed 2025-09-05 by wsnyder