verilator/verilator #2124: Document --xml fields

wsnyder, 2020-01-19:
We should document the --xml output format.

One alternative would be a docs/xml.adoc, but perhaps better to make a real XML schema with documentation tags. We can then validate the XML output against the schema.

@pieter3d #2122

pieter3d, 2020-01-19:
I'll look into this

pieter3d, 2020-01-19:
Hmm, so writing a complete schema starts to approach the complexity of the SystemVerilog syntax in BNF form. Is that worthwhile?

wsnyder, 2020-01-20:
Good point, perhaps we just make an .adoc describing the general concept, plus say fl is deprecated, the loc format, and how datatype ids reference later nodes?

wallento, 2020-01-20:
Hi, I have a dtd on my disk. Is this useful?

wallento, 2020-01-20:
Oh, sorry, I have to take it back. Seems I deleted the branch (accidentally perhaps..)

wsnyder, 2020-01-23:
Closed by #2136

closed 2020-01-23 by wsnyder