verilator/verilator #2126: Translate Verilog $dump* task to internal VCD dumper

agrobman, 2020-01-20:
to increase compatibility with verilog simulators, provide translation of
$dump* tasks  to verilator internal VCD dumper functions.

wsnyder, 2020-01-20:
At present Verilator has the dump stuff external because it's supported and reasonable to have multiple Verilated designed instantiated in the same C++ testbench.  So the question arises how does $dumpfile interact?  Probably the most logical proposal is it is global (cross-design) somewhat like $finish.

May also be interactions with @toddstrader 's protect code.

Also there's FST dumping.  Probably $dump* should do either FST or VCD, whichever was specified at Verilation time.

wsnyder, 2020-03-02:
Added support for $dumpfile in git towards 4.030.  See documentation for how to use this.

toddstrader, 2020-03-02:
Looks like two tests that were added with this commit broke --vltmt:
https://travis-ci.com/verilator/verilator/jobs/292820704
@agrobman can you take a look?  I'm not sure if these tests are even expected to work when multi-threaded.

agrobman, 2020-03-02:
Todd,

I have to way to test anything with verilator later than 4.020,
I have no  permissions to commit anything to the vault ..

Alex

Looks like two tests that were added with this commit broke --vltmt:
https://travis-ci.com/verilator/verilator/jobs/292820704
@agrobman<https://github.com/agrobman> can you take a look? I'm not sure if these tests are even expected to work when multi-threaded.

toddstrader, 2020-03-02:
@agrobman, ah sorry.  I thought the commit was authored by you.  @wsnyder same question about the new tests.

wsnyder, 2020-03-02:
Fixed, sorry took a few attempts as threads seem to behave differently on travis' VMs.

closed 2020-03-02 by wsnyder