verilator/verilator #2127: generate tb.cpp TB wrapper file automatically

agrobman, 2020-01-20:
Would be nice if the C++ TB wrapper file is generated automatically by extracting clock generators of the verilog TB to it ... Or at least suggest it in some basic form/template. .

wsnyder, 2020-01-20:
Once #236 which will allow clocking is done, we should be able to make the top level.

Until then, unfortunately the examples is the best template as there's no way to intuit the clocks and stimulus.

wallento, 2020-01-21:
For people wanting a generic VPI toplevel (stimuli and clocking generated from VPI), cocotb has a good starting point: https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/verilator/verilator.cpp

wsnyder, 2020-12-13:
Verilator will generate a top level testbench using "--main"; using "--main --exe --build" will do everything to make the model. --main is only experimental so undocumented, as it isn't usable for most designs use until delays are supported (#236).

wsnyder, 2022-08-16:
--main will be advertised as part of v5.

closed 2022-08-16 by wsnyder