verilator/verilator #2142: Compillation breaks: assigning to 'CData' (aka 'unsigned char') from incompatible type 'CData [3]'

yurivict, 2020-01-29:
```
./Vtest_bench.cpp:1470:19: error: assigning to 'CData' (aka 'unsigned char') from incompatible type 'CData [3]'
        = vlTOPp->test_bench__DOT____Vcellinp__x____pinNumber3;
          ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I couldn't minimize a testcase.

verilator-4.026

wsnyder, 2020-01-30:
Look for something where pin3 on your testbench is connecting an array or wide to non-array or something like that.  Also try -Oi.  This might be a case where the code is bad but an error is missing, see if it runs in other simulators.

yurivict, 2020-01-30:
I found the mistake. It was that a wire with a mismatching dimension was connected to the module. I just couldn't reduce the case.

wsnyder, 2020-03-01:
Really should be reported as a lint error, but reporting similar problems is already on the list.

closed 2020-03-01 by wsnyder