verilator/verilator #2143: genvar causes errors: error: no member named ...

yurivict, 2020-01-29:
The attached testcase fails:
```
In file included from Vtest_bench__ALLcls.cpp:3:
./Vtest_bench.cpp:144:33: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
        = ((2U >= (3U & vlTOPp->test_bench__DOT__x__DOT__nnn)) 
                        ~~~~~~  ^
./Vtest_bench.cpp:146:27: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
           [(3U & vlTOPp->test_bench__DOT__x__DOT__nnn)]);
                  ~~~~~~  ^
./Vtest_bench.cpp:160:13: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
    vlTOPp->test_bench__DOT__x__DOT__nnn = 3U;
    ~~~~~~  ^
./Vtest_bench.cpp:216:33: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
        = ((2U >= (3U & vlTOPp->test_bench__DOT__x__DOT__nnn)) 
                        ~~~~~~  ^
./Vtest_bench.cpp:218:27: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
           [(3U & vlTOPp->test_bench__DOT__x__DOT__nnn)]);
                  ~~~~~~  ^
./Vtest_bench.cpp:232:30: error: no member named 'test_bench__DOT__x__DOT__nnn' in 'Vtest_bench'
    if ((2U >= (3U & vlTOPp->test_bench__DOT__x__DOT__nnn))) {
                     ~~~~~~  ^
```

Please run the 'run' script in the attached testcase.
The ```genvar``` statement there seems valid.

[bug-missing-genvar-nnn.tar.gz](https://github.com/verilator/verilator/files/4127023/bug-missing-genvar-nnn.tar.gz)

wsnyder, 2020-01-30:
Thanks for your report, the problem was because the code is illegal but wasn't reported. The for loop isn't a generate for, it needs to be "up" above the always.  Added a new warning:

    %Error: t/t_genvar_for_bad.v:22: Genvar not legal in non-generate for (IEEE 2017 27.4): 'nnn'
                                : ... Suggest move for loop upwards to generate-level scope.
          for (nnn=0; nnn<N; nnn=nnn+1) begin
                ^~~

closed 2020-01-30 by wsnyder

yurivict, 2020-01-30:
The error persists after moving the ```for``` loop up above ```always```.

wsnyder, 2020-01-30:
Worked for me, but I see what I think you did, you just moved the for without covering the next always which it also needs to.  I'll fix the error message, but the right code does pass:

```
  generate
      for (nnn=0; nnn<N; nnn=nnn+1)
        begin
           always @(posedge clk)
             begin
                if (oo[nnn] == 0)
                  begin
                     ov[nnn] <= iv[nnn];
                  end
             end
           always @(posedge cc[nnn])
             begin
                oo[nnn] = 1'b0;
             end
        end
endgenerate

```

yurivict, 2020-01-30:
This was exactly the problem. Thank you!