verilator/verilator #2145: always @(...bit selector...) is triggered when a bit is not set

yurivict, 2020-01-30:
This module:
```
module m(sel, i1,i2,i3,i4, o1);

input[3:0] sel;
input i1, i2, i3, i4;
output o1;
reg o1;

always @(sel[0])
begin
        $display("m: sel=%b %b %b %b %b CASE0 triggered i1=%x", sel, sel[0], sel[1], sel[2], sel[3], i1);
        o1 = i1;
end

always @(sel[1])
begin
        $display("m: sel=%b %b %b %b %b CASE1 triggered i2=%x", sel, sel[0], sel[1], sel[2], sel[3], i2);
        o1 = i2;
end

always @(sel[2])
begin
        $display("m: sel=%b %b %b %b %b CASE2 triggered i3=%x", sel, sel[0], sel[1], sel[2], sel[3], i3);
        o1 = i3;
end

always @(sel[3])
begin
        $display("m: sel=%b %b %b %b %b CASE3 triggered i4=%x", sel, sel[0], sel[1], sel[2], sel[3], i4);
        o1 = i4;
end

endmodule
```

Prints these two types of log lines:
```
m: sel=0001 1 0 0 0 CASE0 triggered i1=1
m: sel=0001 1 0 0 0 CASE2 triggered i3=0
```

The second line shouldn't be printed because sel[2] isn't ```1``` and this block shouldn't be triggered.

wsnyder, 2020-01-30:
Verilator follow the synthesis rules, not simulator event model rules for performance.  An "always @(sig)" is same as e.g. evaluating a normal logic gate, and may be called at any time.  If you want it called only on change you need to use e.g. "always @ (posedge x or negedge x)"

wsnyder, 2020-01-30:
See the manual "synthesis subset" section for more information.

closed 2020-01-30 by wsnyder