verilator/verilator #2148: Implementing $rose and $fell

PeterMonsson, 2020-01-31:
Hi,

I would like to take a stab at implementing $rose and $fell and I have 2 questions?

1) What type of AST node should I use for these? 
2) How do I select the lowest bit of a vector/bit? is AstSelBit enough?

Best Regards
Peter

wsnyder, 2020-01-31:
> 1) What type of AST node should I use for these?

You could either make a new AstRose/AstFell/AstStable/AstChanged, or given they are all handled extremely similar to AstPast, add inside AstPast a member variable of type VPastType enumeration which says which flavor.

> 2) How do I select the lowest bit of a vector/bit? is AstSelBit enough?

At the point you convert these (probably V3Assert?) the AstSelBits will be AstSel, so you should make a AstSel if the nodep->width() > 1.

BTW note these in IEEE F.3.4.4:

      $rose(e,c)     ->   $past(b,1,1,c) !== 1 && b === 1, where b is the LSB of e.
      $fell(e,c)     ->   $past(b,1,1,c) !== 0 && b === 0, where b is the LSB of e.
      $stable(e,c)   ->   $past(e,1,1,c) === e.
      $changed(e,c)  ->   $past(e,1,1,c) !== e.

Verilator doesn't support expression 2 (which is 1 in above so that's ok). Add that too if you want (it would just be an if statement inside the $past flop).

Verilator also doesn't support the clock argument ("c"), either error out on that or add it too, it's the clocking expression when making the $past flop.

wsnyder, 2020-05-13:
@PeterMonsson wondering if you're working on this?  Fine if yes, fine if no, just want to return it to the pool in case someone else wants to plow forward.  Thanks

PeterMonsson, 2020-05-16:
Hi Wilson, 

I have been caught up in work. I am planning on attending to it this summer.

I got stuck in the V3Assert stage. I think that I would switch to V3AssertPre to lower $stable, $rose and $fell into an expression over $past. 

Best Regards
Peter

PeterMonsson, 2020-07-24:
I have taken a stab at it, but I am rusty and I don't understand the following error:

- V3AssertPre.cpp:158:assertPreAll:
%Error: Internal Error: t/t_stable.v:57:19: ../V3Ast.cpp:358: Adding already linked node
-node: VARREF 0x7fffc37d44b0 <e835#> {d57at} @dt=0x7fffc37721d0@(G/w32)  dly0 [RV] <- VAR 0x7fffc37cb730 <e811> {d46ar} @dt=0x7fffc37721d0@(G/w32)  dly0 [VSTATIC]  VAR
   57 |       if ($stable(dly0)) $stop;
      |                   ^~~~
                        ... See the manual and https://verilator.org for more assistance.
- V3Ast.cpp:1121:     Dumping obj_vlt/t_stable/Vt_stable_990_final.tree
- V3StatsReport.cpp:231:statsReport:
%Error: Internal Error: Aborting since under --debug
Aborted (core dumped)
%Error: export VERILATOR_ROOT=/home/pmonsson/verilator
%Error: /home/pmonsson/verilator/bin/verilator_bin_dbg --prefix Vt_stable ../obj_vlt/t_stable/Vt_stable__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_stable -OD --debug-check --comp-limit-members 10 --clk clk -f input.vc \+define\+TEST_OBJ_DIR\=obj_vlt/t_stable --debug --no-skip-identical t/t_stable.v
%Error: Command Failed /home/pmonsson/verilator/bin/verilator_bin_dbg --prefix Vt_stable ../obj_vlt/t_stable/Vt_stable__main.cpp --exe --make gmake --x-assign unique -cc -Mdir obj_vlt/t_stable -OD --debug-check --comp-limit-members 10 --clk clk -f input.vc \+define\+TEST_OBJ_DIR\=obj_vlt/t_stable --debug --no-skip-identical t/t_stable.v
%Warning: vlt/t_stable: Exec of perl failed: Starting Verilator 4.039 devel rev v4.026-658-gdc66406e (mod)

The code in question is here:
`    virtual void visit(AstStable* nodep) VL_OVERRIDE {
        if (nodep->sentreep()) return;  // Already processed
        iterateChildren(nodep);
	FileLine* fl = nodep->fileline();
        AstNode* exprp = nodep->exprp()->unlinkFrBack();

	exprp = new AstEq(fl,
			  new AstPast(fl, exprp, NULL),
			  exprp); //new AstVarRef(fl, exprp, true)
	exprp->dtypeSetLogicBool();
        nodep->replaceWith(exprp);
        nodep->sentreep(newSenTree(nodep));
        VL_DO_DANGLING(pushDeletep(nodep), nodep);
    }
`
I think that it may be the dual references to exprp in the AstEq constructor. How do I resolve this? Any pointers will be appreciated.

wsnyder, 2020-07-25:
Right, the second use of exprp should be exprp->cloneTree(), that is make a copy that's identical function.

wsnyder, 2020-07-28:
Stable added in #2488, thanks!  Might you be willing to also look at $rose and $fell?

PeterMonsson, 2020-07-29:
Yes. They should follow the same structure, so hopefully next week.

wsnyder, 2020-08-14:
Thanks for your work on this, finished I believe.

closed 2020-08-14 by wsnyder