verilator/verilator #2150: Implementing $root

Kuree, 2020-02-03:
I notice that `$root` was mentioned in some of the earlier release logs such as [3.864](https://www.veripool.org/news/183-Verilator-Verilator-3-864-Released). However the `master` branch no longer works with `$root`.

Here is a minimal example of code that runs on commercial simulators:
```SystemVerilog
module mod();
integer value = 1;
initial begin
    $display("value is %d", $root.mod.value);
end
endmodule
```

Here is the error I got when running the verilator with lint only option:
```
$ verilator --lint-only mod.sv 
%Error: mod.sv:4: Unsupported: SystemVerilog 2005 reserved word not implemented: '$root'
    $display("value is %d", $root.mod.value);
                            ^~~~~
%Error: mod.sv:4: syntax error, unexpected '.', expecting TYPE-IDENTIFIER
    $display("value is %d", $root.mod.value);
                                 ^
%Error: Exiting due to 2 error(s)
        ... See the manual and https://verilator.org for more assistance.
```

For a quick reference, `$root` is defined in 1800-2017 23.3.1, so this should be a IEEE feature.

Although I never touched Verilator internal before, I'm very interested in helping implementing this feature. Please let me know if there is a place I should start looking at.

wsnyder, 2020-02-03:
Great, would love to have the assistance to add this.  I think the $root reference in the change log was just a way of saying there was two top conflicts not that $root itself worked.

The first step is to make a test_regress format test (see the internals doc), it would be good to make sure there is a e.g, "logic mod;" inside a function so that $root.mod is required to disambiguate which mod is needed.

To support this, add $root to verilog.l replacing the error you see.  Add the token to verilog.l, there's comments where it properly goes.  Add a new AstRoot in AstNodes.h, and in verilog.y make a AstRoot. Search for where AstDot is used; ultimately the code uses AstDot (the "." in a dotted id) to find the symbols in an AstSym; in the case of $root rather than starting the symbol search in the normal place, it needs to use the root symbol tree, then search down for each dotted component from there as normal.

wsnyder, 2020-05-02:
Fixed in git towards 4.034.

closed 2020-05-02 by wsnyder