verilator/verilator #2152: Creating testbench.cpp

hounaidarekik, 2020-02-04:
Hi,
I am working in a project and I want to create my testbench for the Core module.
here all my steps:
1) to generate VCore.cpp, i type the following command:
`verilator -Wall -cc Core.v --exe sim_main.cpp `
2) to link testbench with VCore, i type the following command:
`make -j -C obj_dir -f VCore.mk`
3) to execute VCore, i type the following command:
`./VCore --maxcycles=2000000 --ispm=/home/hounaida.rekik/Desktop/Flexpret3/flexpret-master/tests/Hello/build/emulator/hello.inst.mem --dspm=/home/hounaida.rekik/Desktop/Flexpret3/flexpret-master/tests/Hello/build/emulator/hello.data.mem >/home/hounaida.rekik/Desktop/Flexpret3/flexpret-master/tests/Hello/results/hello.out 2>&1`
I didn't succeed to open hello.inst.mem file.

here is VCore.h:
```
#ifndef _VCore_H_
#define _VCore_H_

#include "verilated_heavy.h"

class VCore__Syms;

//----------

VL_MODULE(VCore) {
  public:
    
    // PORTS
    // The application code writes and reads these signals to
    // propagate new values into/out from the Verilated model.
    VL_IN8(clock,0,0);
    VL_IN8(reset,0,0);
    VL_IN8(io_imem_enable,0,0);
    VL_IN8(io_imem_write,0,0);
    VL_OUT8(io_imem_ready,0,0);
    VL_IN8(io_dmem_enable,0,0);
    VL_IN8(io_dmem_byte_write_0,0,0);
    VL_IN8(io_dmem_byte_write_1,0,0);
    VL_IN8(io_dmem_byte_write_2,0,0);
    VL_IN8(io_dmem_byte_write_3,0,0);
    VL_OUT8(io_bus_enable,0,0);
    VL_OUT8(io_bus_write,0,0);
    VL_IN8(io_gpio_in_3,0,0);
    VL_IN8(io_gpio_in_2,0,0);
    VL_IN8(io_gpio_in_1,0,0);
...............................................
    struct {
...................................................
	VL_SIG(Core__DOT__imem__DOT__ispm[4096],31,0);
	VL_SIG(Core__DOT__dmem__DOT__dspm[4096],31,0);
    };

```

Here is my testbench.cpp:
```
#include "VCore.h"
#include "/usr/local/share/verilator/include/verilated.h"
#include <iostream>
#include <getopt.h> // For getopt_long().
    VCore *top;                      // Instantiation of module

      vluint64_t main_time = 0;       // Current simulation time
        // This is a 64-bit integer to reduce wrap over issues and
        // allow modulus.  You can also use a double, if you wish.

        double sc_time_stamp () {       // Called by $time in Verilog
            return main_time;           // converts to double, to match
                                        // what SystemC does
        }

        int main(int argc, char** argv) {
            Verilated::commandArgs(argc, argv);   // Remember args
                bool done = false;
    bool fail = false;
    uint64_t cycle = 0;
    uint64_t max_cycles = 0;
    const char *ispm_filename = NULL;
    const char *dspm_filename = NULL;
    bool vcd = false;
    FILE *vcd_file = NULL;
    const char *vcd_filename = "trace.vcd";
    int vcd_start = 0;

    // Parse command line options.
    int current_option;
    int option_index = 0;
    static struct option long_options[] = {
        {"sweep", no_argument, &sweep, 1},
        {"trace", no_argument, &trace, 1}, // Text trace.
        {"maxcycles", required_argument, 0, 'm'},
        {"ispm", required_argument, 0, 'i'},
        {"dspm", required_argument, 0, 'd'},
        {"vcd",  optional_argument, 0, 'v'}, // VCD file.
        {"vcdstart",  optional_argument, 0, 's'}, // VCD start cycle.
        {0, 0, 0, 0}
    };

    while((current_option = getopt_long(argc, argv, "i:d:vs", long_options, &option_index)) != -1) {
        switch(current_option) {
            case 0:
                break;
            case 'm':
                max_cycles = atoi(optarg);
                printf("maxcycles: %ld\n", max_cycles);
                break;
            case 'i':
                ispm_filename = optarg;
                printf("ispm: %s\n", ispm_filename);
                break;
            case 'd':
                dspm_filename = optarg;
                printf("dspm: %s\n", dspm_filename);
                break;
            case 'v':
                vcd = true;
                if(optarg != NULL) {
                  vcd_filename = optarg;
                }
                printf("vcd: %s\n", vcd_filename);
                break;
            case 's':
                if(optarg != NULL) {
                    vcd_start = atoi(optarg);
                    printf("vcdstart: %d\n", vcd_start);
                }
                break;
            default:
                return -1;
        }
    }
    // Open VCD trace dump file (if enabled).
    if(vcd) {
        vcd_file = fopen(vcd_filename, "w");
    }
            top = new VCore;             // Create instance
            top->reset = 0;           // Set some input
            while (!Verilated::gotFinish()) {
                if (main_time > 10) {
                    top->reset = 1;   // Deassert reset
                }
                if ((main_time % 10) == 1) {
                    top->clock = 1;       // Toggle clock
                }
                if ((main_time % 10) == 6) {
                    top->clock = 0;
                }
                top->eval();            // Evaluate model
                  // Initialize memories.
                top->Core__DOT__imem__DOT__ispm.read_hex(ispm_filename);
                main_time++;            // Time passes...
            }

            top->final();               // Done simulating
            //    // (Though this example doesn't get here)
            delete top;
            

}
```
i have a problem with this line of code `top->Core__DOT__imem__DOT__ispm.read_hex(ispm_filename);`
Any idea how can i write my testbench in cpp ??? how can i read `VL_SIG(Core__DOT__imem__DOT__ispm[4096],31,0);` ??? am i in the right way ?? there is a useful links for this??

wsnyder, 2020-02-04:
Avoid peeking into the model internals (anything with DOT). Just use $readmemh in your verilog.  If this operation needs to be triggered from C++, then use a DPI export call.

hounaidarekik, 2020-02-04:
Thanks for your help.
In fact, i obtain verilog code after compiling chisel3 code by sbt compiler.
Here the command:
` java -Xmx1024M -Xss8M -XX:MaxPermSize=128M -jar sbt-launch.jar "project Core" "run 1c-4tf-128i-64d`
I check th Core.V code and i find the following sequence:
```
_RAND_0 = {1{$random}};
  `ifdef RANDOMIZE_MEM_INIT
  for (initvar = 0; initvar < 4096; initvar = initvar+1)
    ispm[initvar] = _RAND_0[31:0];
  `endif // RANDOMIZE_MEM_INIT
```
So my question is : should i modify the obtained verilog code (adding $readmemh) ??

wsnyder, 2020-02-04:
Sorry, I think this is a question for chisel, not Verilator.

hounaidarekik, 2020-02-04:
Sorry,I have another question;Could it be a problem of `verilator -Wall -cc Core.v --exe sim_main.cpp` that i had use it and i should use `verilator -Wall -sc Core.v`  to get the access to the # bus? 
Thanks for your help

wsnyder, 2020-02-05:
-sc is if you are using SystemC. Doesn't look like you are.

hounaidarekik, 2020-02-06:
Hi,
Sorry, but i don't understand why when i run verilator --cc Core.v i obtain in VCore.cpp the top module declared in this way "VL_MODULE(VCore)" and the # description and in the # examples i find "SC_MODULE".  What's the difference between the 2 modules??
will "VL_MODULE" module  plug directly into a SystemC?
Thanks for your help

wsnyder, 2020-02-06:
In the top, SC_MODULE is a SystemC module, VL_MODULE is non-systemC, you must use the right one depending on if you want SystemC or not (-sc or -cc option to Verilator).  The Verilator examples I am 100% sure are consistent with this or they wouldn't compile.

hounaidarekik, 2020-02-06:
Yes, i understand.
But if i want to use $readmemh in verilog and DPI interface in c++, should i switch to -sc option or i can use it even with -cc ? 
Tahnks a lot for your heeeeeeeeeeeeellpp

wsnyder, 2020-02-06:
$readmem has nothing to do with SystemC or not, so use --cc.

hounaidarekik, 2020-02-07:
So, as you said for me to initialize memory, iuse the $readmemh in verilog as it's shown in the following code:
```
reg [31:0] ispm [0:4095]; // @[ispm.scala 42:20:@2106.4]
export "DPI-C" task readh;
task readh;
$readmemh("/home/hounaida.rekik/Desktop/Flexpret3/flexpret-master/tests/Hello/build/emulator/hello.data.mem", dspm);
endtask
```
and in the testbench cpp, i add the following lines:
`VCore__Dpi.h` and
```
c->readh();
```
But i get the following error:
```
%Error: unknown:0: Testbench C called 'readh' but scope wasn't set, perhaps due to dpi import call without 'context'
Aborting...
```
I can't see where is the mistake. shoul i use $readmemh in different way ??

wsnyder, 2020-02-07:
See IEEE 1800-2017 section 35.5.3 "Context tasks and functions." You need to use
svSetScope.  I'll have the message mention that.

wsnyder, 2020-02-08:
Think is answered.

closed 2020-02-08 by wsnyder