verilator/verilator #2153: Docker container not returning actual exit code

SmallRoomLabs, 2020-02-06:
When running verilator in docker (from hub.docker.com/r/verilator/verilator) the exit code is always set to 1 regardless success or not. It seems like the wrapper script is the culprit. (I know I can override the entrypoint and run the executable directly without the wrapper, but it'd be nice to always have the proper exit code by default)

With these two test files
```
MatsMBP:status mats$ cat ok.v
module top(input a, output y);
assign y=~a;
endmodule

MatsMBP:status mats$ cat err.v
module top(input a, output y);
assign y=~b;
endmodule
```

I get the correct exit codes when I run a local installation at my laptop.

```
MatsMBP:status mats$ /usr/local/Cellar/verilator/4.020/bin/verilator_bin --lint-only ok.v ; echo $?
0

MatsMBP:status mats$ /usr/local/Cellar/verilator/4.020/bin/verilator_bin --lint-only err.v ; echo $?
%Error: err.v:2: Can't find definition of variable: 'b'
assign y=~b;
          ^
%Error: Exiting due to 1 error(s)
10
```

But when running in the container I always get 1 as the exit code

```
MatsMBP:status mats$ docker run --rm -w/work -v${PWD}/:/work verilator/verilator --lint-only ok.v ; echo $?
1

MatsMBP:status mats$ docker run --rm -w/work -v${PWD}/:/work verilator/verilator --lint-only err.v ; echo $?
%Error: err.v:2: Can't find definition of variable: 'b'
assign y=~b;
          ^
%Error: Exiting due to 1 error(s)
1
```

wsnyder, 2020-02-06:
@wallento

wsnyder, 2020-03-29:
Fixed in git.

closed 2020-03-29 by wsnyder