verilator/verilator #2154: Compilation error: call of overloaded 'log(uint32_t&)' is ambiguous

brancoliticus, 2020-02-06:
My system is illumos OpenIndiana rolling release updated to latest version with development packages installed. g++ is at version 6.5.0.

Here is what I did:
git clone https://git.veripool.org/git/verilator
cd verilator
git checkout stable
autoconf
./configure
gmake

This is the error output:

g++ -I/usr/local/include -std=gnu++14  -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG        -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Partition.cpp
../V3Partition.cpp: In static member function 'static uint32_t LogicMTask::stepCost(uint32_t)':
../V3Partition.cpp:547:34: error: call of overloaded 'log(uint32_t&)' is ambiguous
         double logcost = log(cost);
                                  ^
In file included from /usr/include/math.h:35:0,
                 from /usr/gcc/6/include/c++/6.5.0/cmath:45,
                 from ../V3Ast.h:32,
                 from ../V3OrderGraph.h:50,
                 from ../V3Partition.h:28,
                 from ../V3Partition.cpp:30:
/usr/include/iso/math_iso.h:72:15: note: candidate: double std::log(double)
 extern double log(double);
               ^~~
/usr/include/iso/math_iso.h:172:15: note: candidate: float std::log(float)
  inline float log(float __X) { return __logf(__X); }
               ^~~
/usr/include/iso/math_iso.h:215:21: note: candidate: long double std::log(long double)
  inline long double log(long double __X) { return __logl(__X); }
                     ^~~
gmake[2]: *** [../Makefile_obj:286: V3Partition.o] Error 1
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake[1]: *** [Makefile:70: ../bin/verilator_bin_dbg] Error 2
gmake[1]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src'
gmake: *** [Makefile:230: verilator_exe] Error 2

wsnyder, 2020-02-06:
Try 
  double logcost = log(static_cast<double>(cost));

If that works feel free to make a pull request, add your name to docs/CONTRIBUTORS and you can get the credit. :)

brancoliticus, 2020-02-07:
Now it's a different error:

g++ -I/usr/local/include -std=gnu++14  -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG        -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3Partition.cpp
../V3Partition.cpp: In static member function 'static uint32_t LogicMTask::stepCost(uint32_t)':
../V3Partition.cpp:547:41: error: expected '<' before '(' token
         double logcost = log(static_cast(cost));
                                         ^
../V3Partition.cpp:547:41: error: expected type-specifier before '(' token
../V3Partition.cpp:547:41: error: expected '>' before '(' token
gmake[2]: *** [../Makefile_obj:286: V3Partition.o] Error 1
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake[1]: *** [Makefile:70: ../bin/verilator_bin_dbg] Error 2
gmake[1]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src'
gmake: *** [Makefile:230: verilator_exe] Error 2

Side note: I hope this doesn't degenerate into a wack-a-mole type situation with hundreds of errors to fix one compilation error at a time.

Side note 2: It's not my credit so I don't want it.

brancoliticus, 2020-02-07:
I fixed the compilation at least with: double logcost = log(static_cast<double>(cost));

But now I get a new error:

g++ -I/usr/local/include -std=gnu++14  -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG        -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -c ../V3TSP.cpp
../V3TSP.cpp: In member function 'virtual int TspTestState::cost(const TspTestState*) const':
../V3TSP.cpp:552:49: error: call of overloaded 'sqrt(unsigned int)' is ambiguous
         return lround(sqrt(xabs*xabs + yabs*yabs));
                                                 ^
In file included from /usr/include/math.h:35:0,
                 from /usr/gcc/6/include/c++/6.5.0/cmath:45,
                 from ../V3TSP.cpp:35:
/usr/include/iso/math_iso.h:77:15: note: candidate: double std::sqrt(double)
 extern double sqrt(double);
               ^~~~
/usr/include/iso/math_iso.h:183:15: note: candidate: float std::sqrt(float)
  inline float sqrt(float __X) { return __sqrtf(__X); }
               ^~~~
/usr/include/iso/math_iso.h:232:21: note: candidate: long double std::sqrt(long double)
  inline long double sqrt(long double __X) { return __sqrtl(__X); }
                     ^~~~
gmake[2]: *** [../Makefile_obj:286: V3TSP.o] Error 1
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake[1]: *** [Makefile:70: ../bin/verilator_bin_dbg] Error 2
gmake[1]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src'
gmake: *** [Makefile:230: verilator_exe] Error 2

wsnyder, 2020-02-07:
Same sort of fix, might need several more....

return lround(sqrt(static_cast<double>(xabsxabs + yabsyabs)));

brancoliticus, 2020-02-07:
Please note that `static_cast<double>` gets printed as static_cast when not inside github interface code markdown. So I presume you wrote `static_cast<double>` and the github interface ate the `<double>` part?

Did you therefore write return `lround(sqrt(static_cast<double>(xabs*xabs + yabs*yabs)));` instead of `return lround(sqrt(static_cast(xabsxabs + yabsyabs)));` as I see?

wsnyder, 2020-02-07:
Sorry, should have added spaces (edited above).

           return lround(sqrt(static_cast<double>(xabsxabs + yabsyabs)));

There aren't any uses of "long double" as Verilog doesn't support that type.

brancoliticus, 2020-02-07:
I hit a perl issue:

pod2man bin/verilator verilator.1
gmake: pod2man: Command not found
gmake: *** [Makefile:271: verilator.1] Error 127

I have a pod2{man,text,html} at `/opt/local/bin/pod2man`. How do I tell gmake to use them? I also don't have pod2latex.

brancoliticus, 2020-02-07:
I changed PATH=$PATH:/opt/local/bin and it finished the compilation.
Now gmake test fails.

Output from 'gmake test>~verilator_gmake_test_output.txt' :

------------------------------------------------------------
making verilator in src
gmake -C src 
gmake[1]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src'
gmake -C obj_dbg -j 1  TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj serial
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake[2]: Nothing to be done for 'serial'.
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake -C obj_dbg       TGT=../../bin/verilator_bin_dbg VL_DEBUG=1 -f ../Makefile_obj
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
      Compile flags:  g++ -I/usr/local/include -I/usr/local/include -std=gnu++14 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj serial_vlcov
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake[2]: Nothing to be done for 'serial_vlcov'.
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake -C obj_dbg       TGT=../../bin/verilator_coverage_bin_dbg VL_DEBUG=1 VL_VLCOV=1 -f ../Makefile_obj
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
      Compile flags:  g++ -I/usr/local/include -I/usr/local/include -std=gnu++14 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_dbg'
gmake -C obj_opt -j 1  TGT=../../bin/verilator_bin -f ../Makefile_obj serial
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_opt'
gmake[2]: Nothing to be done for 'serial'.
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_opt'
gmake -C obj_opt       TGT=../../bin/verilator_bin -f ../Makefile_obj
gmake[2]: Entering directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_opt'
      Compile flags:  g++ -I/usr/local/include -I/usr/local/include -std=gnu++14 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG -O2 -MP -Wno-unused-parameter -Wno-shadow -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
gmake[2]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src/obj_opt'
gmake[1]: Leaving directory '/export/home/branco/go/src/verilator_from_source/verilator/src'
test_regress/t/t_a1_first_cc.pl
======================================================================
dist/t_a1_first_cc: ==================================================
dist/t_a1_first_cc: %Skip: Skip: scenario 'dist' not enabled for test
======================================================================
vlt/t_a1_first_cc: ==================================================
	perl ../bin/verilator --debug --debugi 0 --gdbbt --no-dump-tree -V
-n
No stack.
Starting Verilator 4.026 2020-01-11 rev v4.026-2-g0c6c83e2
Starting Verilator 4.026 2020-01-11 rev v4.026-2-g0c6c83e2
Verilator 4.026 2020-01-11 rev v4.026-2-g0c6c83e2

Copyright 2003-2020 by Wilson Snyder.  Verilator is free software; you can
redistribute it and/or modify the Verilator internals under the terms of
either the GNU Lesser General Public License Version 3 or the Perl Artistic
License Version 2.0.

See https://verilator.org for documentation

Summary of configuration:
  Compiled in defaults if not in environment:
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /usr/local/share/verilator

Environment:
    PERL               = 
    SYSTEMC            = 
    SYSTEMC_ARCH       = 
    SYSTEMC_INCLUDE    = 
    SYSTEMC_LIBDIR     = 
    VERILATOR_ROOT     = /export/home/branco/go/src/verilator_from_source/verilator/test_regress/..
    VERILATOR_BIN      = 
[Thread debugging using libthread_db enabled]
[Inferior 1 (process 1549    ) exited normally]
No stack.
[... cut ...]
vlt/t_a1_first_cc: %Error: Exec of make failed: make: Warning: Can't find `.mk': No such file or directory
vlt/t_a1_first_cc: FAILED: Exec of make failed: make: Warning: Can't find `.mk': No such file or directory

======================================================================
	#vlt/t_a1_first_cc: %Error: Exec of make failed: make: Warning: Can't find `.mk': No such file or directory
		make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Unsup 0  Time 0:04

wsnyder, 2020-02-07:
Looks like "make" isn't GNU make.  Run whatever command gets you gnu make, perhaps gmake.

brancoliticus, 2020-02-07:
I did run 'gmake test'. It seems like usage of 'make' is embedded in the script, not the command I used to launch it.

wsnyder, 2020-02-07:
Please pull and use master "git pull ; git checkout master" where you should now be able to "export MAKE=gmake" then this should work.  If not let's see what's next.

BTW we won't be officially regressing OpenSolaris/illumos. By using that, you should expect you mayl need to do some small amount of occasional debug/maintenance and pull requests to keep it working.

brancoliticus, 2020-02-08:
I did:
`git stash
git pull
git checkout master
git statsh pop
PATH=$PATH:/opt/local/bin
gmake
MAKE=gmake
make test`

The result is: 
`make: Fatal error in reader: Makefile, line 204: Unexpected end of line seen`

wsnyder, 2020-02-08:
gmake MAKE=gmake test

brancoliticus, 2020-02-08:
gmake MAKE=gmake test >~/verilator_gmake_test.txt 2>~/verilator_gmake_test_error_output.txt

didn't work.

[verilator_gmake_test.txt](https://github.com/verilator/verilator/files/4174778/verilator_gmake_test.txt)
[verilator_gmake_test_error_output.txt](https://github.com/verilator/verilator/files/4174780/verilator_gmake_test_error_output.txt)

wsnyder, 2020-02-08:
You don't have the fixes, you need to "git pull".

brancoliticus, 2020-02-08:
git log:

commit 70358e8587fec91d7597d560d162fd6a9d39183c (HEAD -> master, origin/master, origin/HEAD)
Author: Wilson Snyder <wsnyder@wsnyder.org>
Date:   Sat Feb 8 10:58:07 2020 -0500

    Fix compiler warning.

`gmake test >~/verilator_gmake_test.txt 2>~/verilator_gmake_test_error_output.txt`:

[verilator_gmake_test.txt](https://github.com/verilator/verilator/files/4174886/verilator_gmake_test.txt)
[verilator_gmake_test_error_output.txt](https://github.com/verilator/verilator/files/4174889/verilator_gmake_test_error_output.txt)

wsnyder, 2020-02-08:
This

	make -C obj_vlt/t_a1_first_cc

Should be saying gmake.  Not sure why it isn't now, please debug and provide whatever patches make it work.

wsnyder, 2020-02-24:
Assuming this is now working, otherwise please reopen.

closed 2020-02-24 by wsnyder

brancoliticus, 2020-03-07:
Finally I got around to this.

It seems for the longest time I was running `gmake test` instead of `gmake MAKE=gmake test` as you said. Sorry for that.

`gmake MAKE=gmake test > verilator_gmake_test_output 2> verilator_gmake_test_error_output`:
[verilator_gmake_test_output.txt](https://github.com/verilator/verilator/files/4302490/verilator_gmake_test_output.txt)
[verilator_gmake_test_error_output.txt](https://github.com/verilator/verilator/files/4302491/verilator_gmake_test_error_output.txt)