verilator/verilator #2158: build in mac failed

quantrpeter, 2020-02-08:
hi, i failed to build in mac. Any hints? thanks Peter

```
/Users/peter/Downloads/verilator/verilator>gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.8)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
/Users/peter/Downloads/verilator/verilator>
/Users/peter/Downloads/verilator/verilator>
/Users/peter/Downloads/verilator/verilator>g++ --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.8)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

```
/opt/local/bin/perl ../flexfix V3Lexer <V3Lexer_pregen.yy.cpp >V3Lexer.yy.cpp
g++ -I/usr/local/include -std=gnu++14  -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow -I/usr/local/include -MMD -I. -I.. -I.. -I../../include -I../../include -DYYDEBUG 	 -ggdb -DVL_DEBUG -D_GLIBCXX_DEBUG -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-undefined-bool-conversion -Wno-shadow -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/usr/local/share/verilator\" -Wno-char-subscripts -Wno-null-conversion -Wno-parentheses-equality -Wno-unused -c ../V3ParseLex.cpp
In file included from ../V3ParseLex.cpp:36:
V3Lexer_pregen.yy.cpp:4136:8: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        yyin.rdbuf(std::cin.rdbuf());
                        ~~~~^
                            ->
V3Lexer_pregen.yy.cpp:4143:9: error: member reference type 'std::ostream *' (aka 'basic_ostream<char> *') is a pointer; did you mean to use '->'?
                        yyout.rdbuf(std::cout.rdbuf());
                        ~~~~~^
                             ->
V3Lexer_pregen.yy.cpp:7962:50: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        YY_CURRENT_BUFFER_LVALUE->yy_input_file = yyin.rdbuf();
                                                                  ~~~~^
                                                                      ->
V3Lexer_pregen.yy.cpp:8081:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type,
      std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin ? arg_yyin->rdbuf() : std::cin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:8082:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type,
      std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout ? arg_yyout->rdbuf() : std::cout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:8084:2: error: use of undeclared identifier 'ctor_common'
        ctor_common();
        ^
V3Lexer_pregen.yy.cpp:8089:14: error: out-of-line definition of 'V3LexerBase' does not match any declaration in 'V3LexerBase'
yyFlexLexer::yyFlexLexer( std::istream& arg_yyin, std::ostream& arg_yyout ):
             ^~~~~~~~~~~
../V3ParseLex.cpp:35:21: note: expanded from macro 'yyFlexLexer'
#define yyFlexLexer V3LexerBase
                    ^~~~~~~~~~~
V3Lexer_pregen.yy.cpp:8080:27: note: type of 1st parameter of member declaration does not match definition ('std::istream *' (aka 'basic_istream<char> *') vs 'std::istream &' (aka 'basic_istream<char> &'))
yyFlexLexer::yyFlexLexer( std::istream* arg_yyin, std::ostream* arg_yyout ):
                          ^
V3Lexer_pregen.yy.cpp:8090:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type,
      std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:8091:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<std::__1::basic_ios<char, std::__1::char_traits<char> >::char_type,
      std::__1::basic_ios<char, std::__1::char_traits<char> >::traits_type> *' (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~
```

wsnyder, 2020-02-08:
Looks like flex is broken, Apple seems to break it differently in every release, you'll need to debug it and perhaps install a different flex and/or fix src/flexfix which hacks around problem.  Let us know what works & patches appreciated if it is something flexfix can repair.

wsnyder, 2020-03-30:
Assume you got it working, closing due to age.

closed 2020-03-30 by wsnyder