verilator/verilator #2159: Is there a way to dump only register values into .vcd?

solomatnikov, 2020-02-09:
Is there a way to dump only register values into .vcd?

There is a tool that calculates combinational signals from register values. Saving only  register values reducing waveform size and improves performance.

wsnyder, 2020-02-09:
What is only a "register"?  You mean something not created by combo logic that is sensitive to everything (as "reg" of course doesn't mean this)?  

What is the tool you are describing?

Anyhow if you are interested in this it would certainly be possible to recognize it and omit such signals.

solomatnikov, 2020-02-09:
Verdi/Siloti

The problem is that in generated Verilog there can be a lot of intermediate comb signals and dumping them all into the waveform is very slow. Verilation is also very slow if dumping of all signals is enabled - sometimes it times out 1 hour limit.

So, the idea is to not dump all comb signals and rely on Verdi/Siloti to re-construct them as needed after simulation.

wsnyder, 2020-02-09:
We'll certainly take a patch if you/someone wants to work in this direction, but IMO the much harder problem is the consuming program and intermediate format.  Ie.g. convince Verdi to open source a format to communicate the needed information, or use GTKwave and likewise improve the information flow.

I agree this is a good direction, but want to close this only because I don't expect progress soon in this area.

wsnyder, 2020-02-09:
P.S. Feel free to continue to discuss - closed doesn't mean closed to comments ;)

closed 2020-02-09 by wsnyder

solomatnikov, 2020-02-09:
There's no need to change anything in Verdi - Verdi reads .vcd and compiles design Verilog separately, so it can trace signal drivers for debugging as long as the signals are in the waveform. This flow already works.

wsnyder, 2020-02-09:
Ah, great, so then a contribution to do this is welcome.