verilator/verilator #2160: .vcd file differ for --threads=1 and --threads=6

oharboe, 2020-02-09:
I would have thought that the .vcd files should be identical, regardless of multithreading. 

If they were, it would be a great way to verify that the simulation result is identical by diffing the .vcd files.

Some differences are innocuos, but they make it hard to compare .vcd files. The "4x" below is for the single threaded case, whereas "`3!" is used in the multithreaded case.

```
   $var wire 64 4x aptos_io_cacheInterface_address [63:0] $end
   $var wire  1 E0 aptos_io_cacheInterface_read $end
```

wsnyder, 2020-02-09:
Some differences may occur due to ordering of multithreading being different, but values should of course match. Use "vcddiff" to compare VCD files.

oharboe, 2020-02-10:
Makes sense.

I figured out how my C++ code broke when I started to use --threads>1, nothing interesting, just a silly bug.

Here's the result from running "vcddiff":

```
 ../../vcddiff/vcddiff ok.vcd ok-threads.vcd
vcddiff: vcddiff.c:135: get_token: Assertion `i < MAXTOKSIZE' failed.
Aborted (core dumped)
```

oharboe, 2020-02-10:
After hacking MAXTOKSIZE to be 1024*1024:

```
 $ vcddiff ok.vcd ok-threads.vcd
TOP.FiddlyBob.io_mainMemory_address[63:0] (h , q.") differs at time 0
	 (?1) (?0) (?1) (?1) (?0) (?1) (?0) (?1) (?0) (?1) (?1) (?1) (?1) (?1) (?1) (?1) (?1) (?1) (?1) (?1) (?0) (?1) (?0) (?1) (?1) (?1) (?0) (?1) (?1) (?1) (?0) (?0) (?1) (?0) (?1) (?0) (?1) (?0) (?0) (?1) (?1) (?1) (?1) (?0) (?0) (?0) (?1) (?1) (?1) (?0) (?0) (?0) (?0) (?1) (?1) (?1) (?0) (?1) (?0) (?0) (?0) (?0) (?0) (?0)
	 (?1) (?0) (?0) (?1) (?1) (?0) (?1) (?0) (?1) (?1) (?1) (?0) (?0) (?1) (?1) (?1) (?0) (?0) (?1) (?1) (?1) (?0) (?1) (?0) (?1) (?1) (?0) (?0) (?1) (?0) (?1) (?1) (?0) (?1) (?0) (?0) (?1) (?0) (?1) (?1) (?1) (?1) (?0) (?0) (?0) (?1) (?0) (?0) (?0) (?1) (?1) (?0) (?0) (?0) (?0) (?0) (?0) (?0) (?0) (?1) (?1) (?0) (?1) (?1)

```

wsnyder, 2020-02-10:
Committed maxsize change to veripool/vcddiff
Thanks for debugging.

closed 2020-02-10 by wsnyder

oharboe, 2020-02-10:
@wsnyder  There's still a difference between --threads 1 and --threads > 1 as incidated by vcddiff. 

Just a thought. I'm testing Chisel generated Verilog, it contains fixed seed random initialization of registers.

Could that cause differences between single and multithreaded?

```
`ifndef SYNTHESIS
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif

```

wsnyder, 2020-02-11:
The threaded and non-threaded codes are optimized differently, they may well declare in a different order. Different init might though is unlikely to make a difference by itself.