verilator/verilator #2161: Warnigs

hounaidarekik, 2020-02-10:
Hi,
To run my verilog code, i use the following command:
`verilator -Wall -cc Core.v`
I get the following warnings:
```
%Warning-VARHIDDEN: Core.v:5990: ... Location of original declaration
%Warning-UNUSED: Core.v:7114: Signal is not used: io_imem_addr
%Warning-UNUSED: Core.v:7115: Signal is not used: io_imem_enable
%Warning-UNUSED: Core.v:7117: Signal is not used: io_imem_write
%Warning-UNUSED: Core.v:7118: Signal is not used: io_imem_data_in
%Warning-UNUSED: Core.v:7120: Signal is not used: io_dmem_addr
%Warning-UNUSED: Core.v:7121: Signal is not used: io_dmem_enable
%Warning-UNUSED: Core.v:7123: Signal is not used: io_dmem_byte_write_0
%Warning-UNUSED: Core.v:7124: Signal is not used: io_dmem_byte_write_1
%Warning-UNUSED: Core.v:7125: Signal is not used: io_dmem_byte_write_2
%Warning-UNUSED: Core.v:7126: Signal is not used: io_dmem_byte_write_3
%Warning-UNUSED: Core.v:7127: Signal is not used: io_dmem_data_in
%Warning-UNUSED: Core.v:7146: Signal is not used: io_int_exts_4
%Warning-UNUSED: Core.v:7147: Signal is not used: io_int_exts_5
%Warning-UNUSED: Core.v:7148: Signal is not used: io_int_exts_6
%Warning-UNUSED: Core.v:7149: Signal is not used: io_int_exts_7
%Warning-UNUSED: Core.v:771: Signal is not driven, nor used: _RAND_0
%Warning-UNUSED: Core.v:773: Signal is not driven, nor used: _RAND_1
%Warning-UNUSED: Core.v:775: Signal is not driven, nor used: _RAND_2
%Warning-UNUSED: Core.v:777: Signal is not driven, nor used: _RAND_3
%Warning-UNUSED: Core.v:779: Signal is not driven, nor used: _RAND_4
%Warning-UNUSED: Core.v:781: Signal is not driven, nor used: _RAND_5
%Warning-UNUSED: Core.v:783: Signal is not driven, nor used: _RAND_6
%Warning-UNUSED: Core.v:785: Signal is not driven, nor used: _RAND_7
%Warning-UNUSED: Core.v:787: Signal is not driven, nor used: _RAND_8
%Warning-UNUSED: Core.v:789: Signal is not driven, nor used: _RAND_9
%Warning-UNUSED: Core.v:791: Signal is not driven, nor used: _RAND_10
%Warning-UNUSED: Core.v:794: Bits of signal are not used: _T_468[2]
%Warning-UNUSED: Core.v:799: Bits of signal are not used: _T_476[2]
%Warning-UNUSED: Core.v:804: Bits of signal are not used: _T_484[2]
%Warning-UNUSED: Core.v:809: Bits of signal are not used: _T_492[2]
%Warning-UNUSED: Core.v:813: Signal is not driven, nor used: _RAND_11
%Warning-UNUSED: Core.v:816: Signal is not driven, nor used: _RAND_12
%Warning-UNUSED: Core.v:830: Signal is not driven, nor used: _RAND_13
%Warning-UNUSED: Core.v:832: Signal is not driven, nor used: _RAND_14
%Warning-UNUSED: Core.v:837: Signal is not driven, nor used: _RAND_15
%Warning-UNUSED: Core.v:848: Signal is not driven, nor used: _RAND_16
%Warning-UNUSED: Core.v:850: Signal is not driven, nor used: _RAND_17
%Warning-UNUSED: Core.v:864: Signal is not driven, nor used: _RAND_18
%Warning-UNUSED: Core.v:866: Signal is not driven, nor used: _RAND_19
%Warning-UNUSED: Core.v:868: Signal is not driven, nor used: _RAND_20
%Warning-UNUSED: Core.v:873: Signal is not driven, nor used: _RAND_21
%Warning-UNUSED: Core.v:875: Signal is not driven, nor used: _RAND_22
%Warning-UNUSED: Core.v:877: Signal is not driven, nor used: _RAND_23
%Warning-UNUSED: Core.v:879: Signal is not driven, nor used: _RAND_24
%Warning-UNUSED: Core.v:881: Signal is not driven, nor used: _RAND_25
%Warning-UNUSED: Core.v:883: Signal is not driven, nor used: _RAND_26
%Warning-UNUSED: Core.v:886: Signal is not driven, nor used: _RAND_27
%Warning-UNUSED: Core.v:889: Signal is not driven, nor used: _RAND_28
%Warning-UNUSED: Core.v:930: Signal is not used: _T_657
%Warning-UNUSED: Core.v:932: Signal is not driven, nor used: _RAND_29
%Warning-UNUSED: Core.v:942: Signal is not used: _GEN_41
%Warning-UNUSED: Core.v:943: Signal is not used: _GEN_42
%Warning-UNUSED: Core.v:944: Signal is not used: _GEN_43
%Warning-UNUSED: Core.v:945: Signal is not used: _GEN_44
%Warning-UNUSED: Core.v:5844: Signal is not driven, nor used: _RAND_0
%Warning-UNUSED: Core.v:5846: Signal is not driven, nor used: _RAND_1
%Warning-UNUSED: Core.v:5848: Signal is not driven, nor used: _RAND_2
%Warning-UNUSED: Core.v:5850: Signal is not driven, nor used: _RAND_3
%Warning-UNUSED: Core.v:5852: Signal is not driven, nor used: _RAND_4
%Warning-UNUSED: Core.v:5854: Signal is not driven, nor used: _RAND_5
%Warning-UNUSED: Core.v:5856: Signal is not driven, nor used: _RAND_6
%Warning-UNUSED: Core.v:5858: Signal is not driven, nor used: _RAND_7
%Warning-UNUSED: Core.v:5860: Signal is not driven, nor used: _RAND_8
%Warning-UNUSED: Core.v:5862: Signal is not driven, nor used: _RAND_9
[... 192 lines cut ...]
%Warning-UNUSED: Core.v:2802: Signal is not used: _GEN_222
%Warning-UNUSED: Core.v:2803: Signal is not used: _GEN_223
%Warning-UNUSED: Core.v:2804: Signal is not used: _GEN_224
%Warning-UNUSED: Core.v:2805: Signal is not used: _GEN_225
%Warning-UNUSED: Core.v:2806: Signal is not used: _GEN_226
%Warning-UNUSED: Core.v:2807: Signal is not used: _T_1179
%Warning-UNUSED: Core.v:2865: Signal is not used: _GEN_293
%Warning-UNUSED: Core.v:2866: Signal is not used: _GEN_294
%Warning-UNUSED: Core.v:2867: Signal is not used: _GEN_295
%Warning-UNUSED: Core.v:2868: Signal is not used: _GEN_296
%Warning-UNUSED: Core.v:2869: Signal is not used: _GEN_298
%Warning-UNUSED: Core.v:2870: Signal is not used: _GEN_299
%Warning-UNUSED: Core.v:2871: Signal is not used: _GEN_300
%Warning-UNUSED: Core.v:2872: Signal is not used: _GEN_301
%Warning-UNUSED: Core.v:2908: Signal is not used: _GEN_341
%Warning-UNUSED: Core.v:2909: Signal is not used: _GEN_342
%Warning-UNUSED: Core.v:2910: Signal is not used: _GEN_343
%Warning-UNUSED: Core.v:2911: Signal is not used: _GEN_344
%Warning-UNUSED: Core.v:2915: Signal is not used: _GEN_350
%Warning-UNUSED: Core.v:2916: Signal is not used: _GEN_351
%Warning-UNUSED: Core.v:2917: Signal is not used: _GEN_352
%Warning-UNUSED: Core.v:2918: Signal is not used: _GEN_353
```
I can't see the error. Any idea about this issue.
Thanks!

seldridge, 2020-02-10:
This looks like Chisel-generate Verilog. Chisel/FIRRTL does not properly guard (with another `ifdef`) the definition of the `_RAND_*` registers. It should do this and this may cut down on some of the warnings you see. Ideally, I'd like to get the Verilog that FIRRTL emits down to zero warnings... (you should be able to use `-Wall -Werror` without problems.)

hounaidarekik, 2020-02-10:
Hi thanks for your help.
Yes , I am working with chisel3 so i obtain two files Core.v and Core.fir when i use sbt compiler then i use the verilator to generate VCore.cpp 
[Core.fir.txt](https://github.com/verilator/verilator/files/4181923/Core.fir.txt)
:
[Core.txt](https://github.com/verilator/verilator/files/4181918/Core.txt)

wsnyder, 2020-02-10:
See link above https://github.com/freechipsproject/firrtl/issues/1372

closed 2020-02-10 by wsnyder