verilator/verilator #2162: Disabling of assert statements

towoe, 2020-02-11:
Since https://github.com/verilator/verilator/commit/83a1bd0675e09c400267847fa5c9df341f360a28 it seems to me that regardless of most* settings, assert statements always terminate the simulation.
\* `+verilator+error+limit+50` will delay the termination.

Previous to the commit I think it was possible to use `Verilated::assertOn(false)` to disable the termination of the simulation based on the firing of any assert statements in the Verilog code.
And in fact they seem to be only activated if the build option `--assert` was used.

Is it possible to ignore all asserts, and if yes, what would be the correct way to do that?

wsnyder, 2020-02-11:
Anything untested will eventually get broken ;)

The intent was not to break assertOn.  However probably assertOn should control only what is under assert(...) etc, and not a standalone $error/$warning.

Could you like to attempt a pull request to fix this?  The V3Assert code needs to add back the assertOn when not under an AstAssert (etc), plus we need some tests.

towoe, 2020-02-11:
Thanks for the quick response.
Will try to fix this and add some tests for it.

commit 18f8cd05 (2020-02-16): Allow assert disable (#2168)

* Add +verilator+noassert flag

This allows to disable the assert check per simulation argument.

* Add AssertOn check for assert

Insert the check AssertOn to allow disabling of asserts.
Asserts can be disabled by not using the `--assert` flag or by calling
`AssertOn(false)`, or passing the "+verilator+noassert" runtime flag.
Add tests for this behavior.
Bad tests check that the assert still causes a stop.
Non bad tests check that asserts are properly disabled and cause no stop
of the simulation.

Fixes #2162.

* Correct file location

* Add description for singl [... cut]
closed 2020-02-16 by wsnyder