verilator/verilator #2165: protected_lib related

fuyong10, 2020-02-12:
Have tried protected_lib, it works!

Two comments:

1. Any way to turn off protection? To instantiate a verilated IP in SOC does not always mean  to stop the visibility inside. To give the VCD file of a full SOC to the IP developer can help the developer debug quickly. Meanwhile, the source code of that IP can still be maintained by the developer.

2. What if an IP has been instantiated several times with different parameters?

toddstrader, 2020-02-12:
Great, I'm glad to hear that it's working for you.

Currently the protected library does no tracing of its own.  My thought for allowing the IP developer to see what's going on is in #1523 (currently being worked on in #2118).  Basically, the end user would generate a trace of what they are doing with the IP and ship the trace (or just the part of the trace which interacts with the IP) to the IP developer who would replay the trace data through their unprotected version of the IP.

However, I gather you're suggesting something else with # 1.  Are you suggesting a mode where the RTL of the IP core will still be protected, but it will trace all of its internal signals without any kind of protection/encryption?  If I'm understanding this correctly, it can't be done today, but this feature could be added in the future.

One significant caveat here is that we protect the signal names in the verilated C++ that will be compiled into the protected library by having --protect-lib imply --protect-ids.  You can certainly override that on the command line so that you get unprotected signal names in the protected library, but that opens the door to more reverse engineering on the compiled IP library.  I'm curious, do proprietary simulators have an analogous feature where they can trace plain text signal names of encrypted RTL?  And is this something IP vendors can opt into when encrypting their RTL?  If so, they'd be susceptible to the same reverse engineering.

Regarding # 2, that's certainly the biggest issue with --protect-lib at present.  #1522 is tracking this, but I'm afraid that will be a pretty long-term project.  If you have a reasonably small set of parameter permutations, my current suggestion would be to ask your IP vendor to provide you with a compiled library for each permutation (i.e. the vendor sets the parameters you supply).

Some additional thoughts on # 2 can be found in the talk I gave at ORConf last year on --protect-lib:
https://www.youtube.com/watch?v=0D2982U84Oo
https://www.veripool.org/papers/protect-lib-orconf-2019.pdf

wsnyder, 2020-02-12:
For 1, note @toddstrader is suggesting to use --protect-lib --no-protect-ids (in that order).

For 2, if you don't mind Verilating multiple times, you could work around this by make a model for each parameterized "flavor"; use the -P argument to Verilator to specify the parameter values, and make sure to use a different model name for each flavor.

toddstrader, 2020-02-12:
> For 1, note @toddstrader is suggesting to use --protect-lib --no-protect-ids (in that order).

This would un-protect the symbols in the protected library (i.e. you can see them with the `strings` command), but it won't add trace data for the protected core in the VCD/FST of the instantiating design.  I believe this is the correct default behavior, namely that protected IP shouldn't expose anything (including scrambled signal names) in the trace data other than its I/O.

However, if this is desirable, I can look at adding an option for the IP vendor --yes-trace-signals-in-the-instantiating-design (but with a better name).  This does intersect with #1518, so we'd have to figure that out first.

fuyong10, 2020-02-13:
thanks Tod and Wilson. Sorry I did not make the suggestion well. My intention is to compile IP(s) without protect, and then integrate the verilated IP(s) in the full SOC. 

The usage scenario is not about the IP from 3rd vendor, rather for the internal design group. Ex:
    FULL_SOC
           IP #(param1, param2..) IP_instance_1 (...)
           IP #(param3, param4..) IP_instance_2 (...)

Assume this is a big SOC, from compile time, Verilator's capability of handling large design, to verilate IPs first, and then instantiate into full SOC. SOC compile only take the reset of design, so might be easier than the full thing. But at run time, hopefully the waveform of full design including the IPs can be generated. 

I re-visited the forum and see quite several threads related to:
1. To use VCS/IUS together with Verilator. I assume DUT is varilated, and testbench with DPI calls are made so connect with VCS/IUS. However, the protect-lib wins since:
      a. protect_lib take care of the DPI and wrapper automatically
      b. the clock analysis etc are done by tools, so does the evaluation way. 
      c. Also a nice thing is protect-lib can support multiple clocks! (correct if I am wrong. I played with tracing_c example by turning sub.v into protect_lib. Module sub has clk and faskclk)
   Seems like there are people use VCS+Verilator successfully.
2.  Embedded IP. This is very close to what I think. I noticed there is a test case,  test_regress/t/emdeded* and will play with it.
3. Somehow remember there is a limitation says, one Verialted IP can be be instanciated by another verilated IP? For example, verilate "IP" mentione above, and then want to verilate "FULL_SOC" with "IP" inside.
4. Also recalled the challenge of handling the comb logic output if the comb logic input is the primary input of the IP. Or needs several eval() here can solve this issue but with the performance downgrade.

Seems like to verilate a FULL SOC with some of its IPs being verilated first is a challenge.

fuyong10, 2020-02-13:
example modifed with tracing_c:

verilator -cc -02 --protect-lib verilated_ip -Mdir obj_dir_ip sub.v
This command works, intention is to verilate module "sub" and put into a lib "verilated_ip"

Add --no-protected-ids:
verilator -cc -02 --protect-lib --no-protected-ids verilated_ip -Mdir obj_dir_ip sub.v
%Error: Cannot find file containing module: verilated_ip

could not find doc for --no-protected-ids, surely my command is incorrect. Any hints here. Thanks,

toddstrader, 2020-02-13:
Changing your command to:
```verilator -cc -02 --protect-lib verilated_ip --no-protect-ids -Mdir obj_dir_ip sub.v```
will at least get you past your current error.  --protect-lib takes an argument (verilated_ip) in this case so you need --no-protect-ids to come after that argument.  Also, you can find --protect-ids in the manual and --no- before a boolean switch just negates the switch.

But yes, if your intention is to use --protect-lib to reduce compile time on a large design you are correct that there are some challenges.  I think your list summarizes things well, but I'm not sure I know what # 3 is referring to.  Perhaps a link to the thread would help here.

Also, even with the above command you won't get signals from the protected library in your top-level trace file.  I modified test_regress/t/t_prot_lib.pl similarly and confirmed this.  The reason is the generated C++ does nothing with respect to tracing.  This might be the simplest change to make in the direction I believe you are trying to go.  If you're interesting in submitting a PR for this or anything else --protect-lib related please let me know and I can try to supply some pointers.

fuyong10, 2020-02-14:
verilator -cc -02 --protect-lib verilated_ip --no-protect-ids -Mdir obj_dir_ip sub.v
=> This command works!
You are right, I did not get the signals. 

OK, my original intention :

    tesbench   <= some complex SV here, so would like to run in VCS or IUS 
         dut       <= verilate dut , and hopefully can have SV/DPI so can be compiled together with testbench in VCS/IUS
              IP1   <= want to verilate as an IP
              IP2    <= want to verilate as an IP

I also learning the dpi_compile_master example. Let me re-think the initial idea and maybe manually make an example and share. Any example that might help are welcomed.

toddstrader, 2020-02-14:
Agreed that an example would be beneficial here.  Also, for additional reference beyond the `examples` directory you may want to look at `test_regress/t/t_prot_lib*` if you haven't done so already.

fuyong10, 2020-02-14:
Thanks tod, will look at the example. 

By the way, have saw several examples. Many if not all are using two phases eval(). In your example, verilated_secret.sv, you have calls of *_combo_update() and *_seq_update(). Well, more eval() in one simulation time can always yield "stable" and more predictable result. But just curious, if Verilator has been schedule the data path statically, will one eval() after clock tick (in *_seq_update) be good enough?  Since verilator is called cycle based, so can I assume that each eval(), regardless of how many input has changes, will always run through all the evaluations, so consume same CPU time?

toddstrader, 2020-02-18:
The reason eval() may be called more than once a cycle is that the testbench (e.g. C++ code) can respond combinatorially.  In other words, you may change your stimulus to the DUT based on something the DUT is doing within the same clock cycle.  This would require you to call eval() again because the DUT hasn't seen the new stimulus.

--protect-lib calls eval() multiple times per cycle for related reasons.  I.e. if there were a combinatorial path through a protected library, we can't wait for a clock edge to evaluate it because that would be too late.

Reducing the number of times eval() is called is on the list of things to do for --protect-lib.

fuyong10, 2020-02-19:
thanks Todd. In Verilator warning message, I learned something like DIDNOTCONVERGE, UNOPTFLAT, GENCLK, IMPERFECTSCH, UNOPT, etc. Maybe I can assume:

1. In  case there is any combination loops inside the DUT( or protected)_ip) itself, Verilator can handle correctly with the generated code inside eval() , means any more evaluation is already part of the eval(). 

2. Only when users suspect that there could be dependencies between protected_ip and rest_of_DUT ( similar to the case described in UNOPT), then users are in charge of do more eval()s instead of only one at clock edge. 

Giving a worst case example:

module protected_ip ( in1, in2, out);  // a simple AND gate 
   assign out = in1 & in2;
endmodule

module DUT ()
 protected_ip IP1 ( in1, in2, out1_2);   // data chain, combo chain
 protected_ip IP2 ( out1_2, in3, out2_3);
 protectde_ip IP3 ( out2_3, in4, out3);

a combinational data chain is formed with several instances of protected_ip. This case, I assume we will need to do mutliple eval()s to propagate data from in1 to out3. Will have quite some impact on performance, yet, needs carefully number of eval() otherwise malfunc ?

toddstrader, 2020-02-19:
The issue you described should be resolved by #1524.  As a workaround, you are correct that you could eval() the DUT multilple times until things converge.

fuyong10, 2020-02-22:
#1524 in github does not have the background context. What I think can be more close is if protect-lib does not enable the encrytion algorithm, but output the SV and DPI directly with the original IP. I recalled you mentioned adding --embedded-ip to serve this purpose in one thread but do not know if that has been done. Just feel it could be a subset of what you have done in protect-lib :-)

toddstrader, 2020-02-24:
I'm guessing you're referring to the discussion in #1572.  There are a few things wrapped up in there.  One is already handled by `--protect-lib foo --no-protect-ids`.  The others are handling parameter variants more gracefully (building each variant independently and wrapping it up in one library) and tracing from the protected library.

If you're looking for either of these features I'd suggest we move the conversation to that issue.  If not, could you create a Verilator test showing exactly what it is you're trying to do?  --protect-lib tests are unfortunately a little more complicated that others, but `test_regress/t/t_prot_lib.pl` should serve as a guide.  It basically does the following:

1. Verilates the protected code with --protect-lib
2. Compiles the verilated C++ into a protected library
3. Compiles (verilates and C++ compiles) the top level including the new protected library
4. Runs the simulation

fuyong10, 2020-02-24:
After re-visit the dpi-compile example from :  https://github.com/toddstrader/dpi-compile, I now have better understand. Guess I was confused by the unencrypted signal names in the VCD file (foo.vcs). 
In the DPI-compile case, I compiled without PROT, but still was able to see a lot of unencrypted signals. Well, those signals are the signals in wrapper (foo.sv). Signals inside the embedded IP (foo_impl.sv) are not there, which should be the expected behave for now.

Think to enable the waveform and debug capability could be the next interesting topic. Since the original purpose was to compile a large design separately, the visibility inside each compiled IP should still be observable. 

Maybe I should close this thread and open a new one?

toddstrader, 2020-02-24:
Correct, I/O should not be hidden or obfuscated as that would make using the protected module impossible.

Agreed that it seems like this issue has run its course.  As you said, please close this one and open a new issue with specifics on what you are trying to do and we'll take it from there.

closed 2020-02-25 by fuyong10