verilator/verilator #2167: Undetected multiple drivers on wire when connected to module output port.

laforest, 2020-02-12:
This seems related to issue #1184 
Given:

parameter WORD_WIDTH = 32; (or any convenient value)
wire [WORD_WIDTH-1:0] foo = 0;

some_module instance_name
(
.output_port (foo)
);

The Verilator linter does not notice the multiple drivers on "foo".
Icarus Verilog raises no warnings, but correctly simulates an X on any bits not set to zero by the module output port.
Vivado synthesis will correctly fail with a multiple drivers error.

This error originated when I changed the type of "foo" from reg to wire, because the driving logic was put into a module, and forgot to remove the intial value.

Perhaps Verilator could check that wires on module output ports do not have any assignments, or that wires with assignments cannot be used to connect to module outputs.

wsnyder, 2020-02-13:
Agreed this should give a MULTIDRIVEN.  Took a quick look as thought would be trivial, but there's some complications so will need to get back to it.

laforest, 2020-02-13:
Thanks. I appreciate it.

codypiersall, 2020-03-05:
I'm looking forward to this change.  Thanks for all the great work on Verilator.  (I'm commenting so GitHub will notify me of updates...)

wsnyder, 2020-03-06:
Note you can also hit "Subscribe" in github without commenting, though your comment is appreciated regardless.

codypiersall, 2020-03-06:
@wsnyder Thanks for the info, much appreciated!  I never noticed that button over on the sidebar.

wsnyder, 2021-02-17:
Complication with this, it's ok and probably shouldn't warn on tristate where 

    assign not_multidriven = 1'bz;  // or something more complicated that sometimes drives
    sub sub (.not_multdriven)   // where this is output pin of instance

closed 2026-11-05 by wsnyder