verilator/verilator #2169: protect_lib and gated clock

fuyong10, 2020-02-14:
For testing gated clock purpose, I modified secret_impl.v under make_protect_lib example:

     always @(posedge clk) begin
changed to:
     wire GC = clk & a[0];
     always @(posedge GC) begin

Compile error. Seems like obj_dir_secret/verilated_secret.sv is not generated correctly:

    // Evaluate clock edges
    always @( ) begin   <==== sensitive list is empty.

toddstrader, 2020-02-14:
Yeah, that definitely sounds like a --protect-lib code generation bug.  Do you want to make a PR for a new test in `test_regression` and/or a fix for this?  If not, I can take a look.

fuyong10, 2020-02-14:
I am quite new to Verilator so really not capable of fixing yet. Please go ahead and take a look.

Ps, what does "PR" here mean?

Millennium-Software-Design, 2020-02-15:
PR: Pull Request.  See https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests

(in the good ol' days we just said, "patch")

toddstrader, 2020-02-18:
Test is in branch:
https://github.com/verilator/verilator/tree/issue-2169

I'll look into a fix next.

toddstrader, 2020-02-20:
As with most things this turns out to be more complicated than I originally anticipated.  I'm still working on the solution, but what's clear to me now is that if you're concerned about performance you'll want to do your clock gating outside of the --protect-lib modules.  See the manual for its advice on clock enables and clocker/clock_enable attributes.

toddstrader, 2020-02-24:
Resolved in #2172.

closed 2020-02-24 by toddstrader