verilator/verilator #2170: verilator fails without providing examples: Unsupported: Blocked and non-blocking assignments to same variable

yurivict, 2020-02-18:
```
%Error-BLKANDNBLK: mmm.v:86: Unsupported: Blocked and non-blocking assignments to same variable: 'test_bench.mmm_ggg.genblk1[6].genblk1[1].mmm_r_s.my_wire'
wire         my_wire;
             ^~~~~~~~~~~~~
                   mmm_ggg.v:107: ... note: In file included from mmm_ggg.v
                   test_bench.v:52: ... note: In file included from test_bench.v
%Error-BLKANDNBLK: mmm.v:86: Unsupported: Blocked and non-blocking assignments to same variable: 'test_bench.mmm_ggg.genblk1[7].genblk1[0].mmm_r_s.my_wire'
wire         my_wire;
             ^~~~~~~~~~~~~
                   mmm_ggg.v:107: ... note: In file included from mmm_ggg.v
                   test_bench.v:52: ... note: In file included from test_bench.v
```

In fact, this wire is always assigned in a non-blocking way, there are only few places.

How can I make verilator say where the blocking assignments are?

wsnyder, 2020-02-18:
Agree that should be better.

markVarUsage in V3Delayed reports this error. Perhaps you could attempt a patch to pass in the AstNodeVarRef* that is portentially going to make the error, and report it?

yurivict, 2020-02-18:
There is no AstNodeVarRef in this function. Where is it supposed to come from?

wsnyder, 2020-02-18:
Probably pass it down from where it is called.

yurivict, 2020-02-19:
I found two lines which make Verilator think that there is a Delayed vs. NonDelayed assignment constraint violation.

Delayed assignment is just ```mywire <= 1;``` which is correct.

But for NonDelayed assignment it shows the output of a mux:
```
mymux  mymux (sel
        , otherwire1
        , otherwire2
        , otherwire3
        , otherwire4
        , otherwire5
        , otherwire6
        , otherwire7
        , otherwire8
        , otherwire9
        , mywire
);
```

The mux doesn't contain any NonDelayed assignments for its output:
```
module mymux  (sel, i1,i2,i3,i4,i5,i6,i7,i8,i9, o1);

input [SBITS:0] sel;
input i1,i2,i3,i4,i5,i6,i7,i8,i9;
output o1;
reg o1

always @(posedge i1 or negedge i1)
begin
	if (sel == 4'b0001)
		o1 <= i1;
end

always @(posedge i2 or negedge i2)
begin
	if (sel == 4'b0010)
		o1 <= i2;
end

always @(posedge i3 or negedge i3)
begin
	if (sel == 4'b0011)
		o1 <= i3;
end

always @(posedge i4 or negedge i4)
begin
	if (sel == 4'b0100)
		o1 <= i4;
end

always @(posedge i5 or negedge i5)
begin
	if (sel == 4'b0101)
		o1 <= i5;
end

always @(posedge i6 or negedge i6)
begin
	if (sel == 4'b0110)
		o1 <= i6;
end

always @(posedge i7 or negedge i7)
begin
	if (sel == 4'b0111)
		o1 <= i7;
end

always @(posedge i8 or negedge i8)
begin
	if (sel == 4'b1000)
		o1 <= i8;
end

always @(posedge i9 or negedge i9)
begin
	if (sel == 4'b1001)
		o1 <= i9;
end

endmodule
```

Also, the problem only came up when this mux grew from 8 to 9 inputs, which is very strange. I think Verilator makes a mistake - there is a bug somewhere that makes Verilator think that there is a NonDelayed assignment there.

yurivict, 2020-02-20:
I think that if you would implement the feature that would always print at least one example of Delayed and NonDelayed assignment lines you would find a bug, because in my case verilator wouldn't be able to find a line for a NonDelayed assignment.

wsnyder, 2020-02-23:
I have an idea what might be wrong, but need a complete test, would you mind please submitting a test that is in the test_regress/t format that shows the error (see docs/internals.adoc for details).

wsnyder, 2020-12-10:
Added examples to warning.

```
%Error-BLKANDNBLK: t/t_order_blkandnblk_bad.v:17:21: Unsupported: Blocked and non-blocking assignments to same variable: 't.array'
   17 |    logic [1:0][3:0] array;
      |                     ^~~~~
                   t/t_order_blkandnblk_bad.v:19:16: ... Location of blocking assignment
   19 |    always_comb array[0] = i;
      |                ^~~~~
                   t/t_order_blkandnblk_bad.v:22:6: ... Location of nonblocking assignment
   22 |      array[1] <= array[0];
      |      ^~~~~
```

closed 2020-12-10 by wsnyder