verilator/verilator #2173: VHDL (mixed-language sim) support

Millennium-Software-Design, 2020-02-21:
How may we help - what is your question?
I think for open source simulators to really take off in the industry they need to have mixed language (VHDL and SystemVerilog) support.  I know very few projects that don't have some "IP" in VHDL even though the main language project language is Verilog/SystemVerilog (or vice versa).

I found a discussion about adding VHDL support from 9 years ago (with Sebastien Van Cauwenberghe).  Did that every go anywhere?  Is there anyone working on this now?

wsnyder, 2020-02-23:
Yes, @svancau has done some work on VHDL as recently as last year, and had some preliminary signs of life. If you would be willing to help out perhaps he can point to his repo and suggest how to help out.

wsnyder, 2020-02-23:
Reopen until @svancau replies.

fredrequin, 2020-02-24:
Hello, just to let you know : I have tried a different approach than @svancau.
Instead of dealing with XML files in NVC and Verilator, I wrote an output filter in NVC that translate VHDL into SystemVerilog files. The first alpha version looks promising, I have already ran with Verilator some VHDL (and mixed HDL) simulation at work.
The repo is here : https://github.com/fredrequin/nvc
You have tou use the --dum-sv option in NVC to generate the SystemVerilog file, something like:
nvc -a foo.vhd --relax=prefer-explicit --dump-sv=foo.sv

closed 2020-03-01 by wsnyder