verilator/verilator #2175: how to enbale VL_DEBUG_IF message

fuyong10, 2020-02-24:
Find VL_DEBUG_IF message could be quite useful to understand the eval() flow, but after the following trial, the message could not print in runtime:

verilator ... -CFLAGS "+DVL_DEBUG'
obj_dir/Vtop +verilator+debug

find several run programs under $VERIALTOR_ROOT/bin, maybe I should try with other program instead of verialtor?

fuyong10, 2020-02-24:
In another small test case, the following works:
verilator ... -CFLAGS "+DVL_DEBUG'
obj_dir/Vtop +verilator+debug

The one that does not show is based on the dpi_compile_master case.

wsnyder, 2020-02-24:
I didn't think GCC accepted "+D" to make a define, it's "-D".  What to look at is the "gcc" line that is printed when you run "make" - see if it has the CFLAGS you requested attached or not.

wsnyder, 2020-03-01:
Assuming you figured this out.

closed 2020-03-01 by wsnyder