verilator/verilator #2176: referencing objects in named generate blocks

agrobman, 2020-02-24:
Hi,

There is a problem to reference memory arrays in demo TB of the swerv EH2/EL2 blocks. 
See email thread below.

> You can use the same RTL, just change your DRAM define as below.
> Verilator seems to add genblk's where other simulators don't,
> unfortunately this isn't a well specified area of the language. Please
> file a bug on Verilator for this to get cleaned up.
>
> diff --git a/testbench/tb_top.sv b/testbench/tb_top.sv
> index 425d711..c9f6287 100644
> --- a/testbench/tb_top.sv
> +++ b/testbench/tb_top.sv
> @@ -686,7 +686,7 @@ end
> endtask
>
> `ifdef VERILATOR
> -`define DRAM(bank)
> rvtop.mem.Gen_dccm_enable.dccm.mem_bank[bank].ram.ram_core
> +`define DRAM(bank)
> rvtop.mem.Gen_dccm_enable.dccm.mem_bank[bank].genblk1.genblk1.genblk1.dccm.dccm_bank.ram_core
> `else
> `define DRAM(bank)
> rvtop.mem.Gen_dccm_enable.dccm.mem_bank[bank].dccm.dccm_bank.ram_core
> `endif
>
>
>
>> Hi Wilson
>>
>> Could you, please, check why do we need `ifdef VERILATOR in the design
>> files
>> as :
>>
>> design/lsu/elx2_lsu_dccm_mem.sv
>>
>> we are accessing memory arrays from TB via full path reference, but 4.20
>> verilator gives an error - can't find it,
>> So I had to pop up parameterizable memory one level up to make it
>> compile
>> ...
>>
>>
>> Thanks
>> Alex

wsnyder, 2020-02-24:
Confirmed.  Basically it looks like other simulators don't add "genblk" to generate blocks without names, which Verilator does.  Need to make sure there's no case where this is needed then fix.

agrobman, 2020-02-25:
All other simulators, I worked with,  do add something like genblkN for every generated unnamed blocks, but exact name and number is not guaranteed. Users have to name the blocks to be able reference something inside them.
The problem here, why verilator adds hierarchy differently than VCS or NCverilog in this specific case. As I mentioned before verilator did compile EH1 , which has one level of generate statement less than our new designs ..
And that’s why I placed the RAM module on the same level as EH1 to make it compile our new designs.

wsnyder, 2020-02-26:
IEEE 2009 added the requirement that directly nested generate blocks not get a genblk name, this had been missed, so basically an "if (x) begin :y else if (x1) begin :z" would end up making genblk1 to represent the implied "else" of the first if.

Fixed in git towards 4.030.  Note this renaming might break some designs assuming the old naming.

closed 2020-02-26 by wsnyder