verilator/verilator #2177: incorrect simulation when a wire used 3 times

smcpeak, 2020-02-25:
I found a case where Verilator's simulation is incorrect.  The bug manifests when a particular circuit uses a wire's value three times, but not if it is only used twice.

The attached [cse-bug.tar.gz](https://github.com/verilator/verilator/files/4248900/cse-bug.tar.gz) contains the minimized reproducer.  When the "sim" program is run, it prints:

```
outercond=1 innercond=0
outercond=0 innercond=1
```

The first line should be logically impossible because outercond is "op2 & addr17 & valid", while innercond is "addr17 & valid", a strictly weaker condition.

From studying the generated C++ code, the problem appears to be that 'innercond' is used before its value is recomputed.  Thus, for most of the circuit, its value is a cycle behind what it should be.  That is why it is 1 in the second line of output, even though for the given circuit inputs on that cycle, it should be 0.

That code generation problem in turn appears related to the fact that the wire 'innercond' is used three times in the circuit.  If I remove one of the uses not involved in printing, then the generated C++ code no longer refers to 'innercond', but instead expands its definition at both of the remaining usage sites, resulting in correct output:

```
outercond=1 innercond=1
outercond=0 innercond=0
```

This bug occurs with all versions I have tested: the current HEAD of master (93ac79981b), v4.028, v4.016, and v4.008.

For ease of reference, here is the reproducer, `DebugModule.v`:

```
module DebugModule(
  input         clock,
  input         valid,
  input  [1:0]  operation,
  input  [6:0]  addr
);
  reg  command_postexec;

  wire  addr17;
  wire  op2;
  wire  op2_and_addr17;
  wire  outercond;
  wire  innercond;
  wire  _GEN_92;

  assign addr17         = addr      == 7'h17;
  assign op2            = operation == 2'h2;
  assign op2_and_addr17 = op2  &  addr17;
  assign outercond      = op2_and_addr17 & valid;
  assign innercond      =         addr17 & valid;

  // Remove this, bug goes away.
  assign _GEN_92        = op2 & innercond;

  always @(posedge clock) begin
    // Remove this block, bug goes away.
    if (innercond) begin
      command_postexec <= valid;
    end

    // This line shows the bug.  On cycle 2, outercode is 1
    // but innercond is 0, which should be impossible.  Then,
    // on the next cycle, 'innercond' shows as 1 even though
    // 'valid' is 0.
    $fwrite(32'h80000002,"outercond=%d innercond=%d\n",outercond,innercond);
  end
endmodule
```

wsnyder, 2020-02-25:
Looks like an ordering bug, I will take a look.  A workaround is to add an eval() between where you set your non-clock inputs and toggle the clock.

smcpeak, 2020-02-26:
Thanks for looking at this!

I can confirm that adding an extra eval() before raising the clock makes the bug go away for this reduced example.  However, my real circuit is of course much more complicated, and I'm concerned that if the generated code has reads before writes for multiple signals, there might be a dependency among them that requires multiple extra eval() calls to fully resolve.  I'm also concerned about the possibility that a single eval() after raising the clock might not be enough.  Since I've already got my hands full debugging the circuit when there are no simulation issues (it's a mess and I didn't write most of it), I'm reluctant to plow ahead with an uncertain foundation.  Do you have high confidence that a single extra eval() before raising the clock is always sufficient?

As this issue is more or less blocking my progress on my present project, another tack would be for me to try to fix it myself, although I haven't spent any time with the Verilator internals.  If you have any tips or hunches about where to look, that might be helpful.  For example, the number 3 seems to be a relevant threshold, apparently controlling whether common subexpressions use named temporaries or are recomputed each time.  Any idea where that threshold comes from?  Raising it to infinity might be sufficient for me in the short term, as simulation performance is not critical at the moment.

wsnyder, 2020-02-26:
I've been looking at fixing this and think for now it is going to have to be expected behavior (which will get documented) at least for the medium term.

To fix this would be complicated and greatly slow down simulations for models that call eval the way that is currently expected by Verilator (and SystemC).

Basically the requirement is/will be all signals that are not clocks are expected to be stable ("evaled") before you do an eval with a clock transition.  Think of this as your non-clock signals need to come from a flop. One way to do this is set all inputs other than clocks into the model on the negedge, or *after* the eval of the previous posedge.  Note this is exactly what you would get by attaching two verilated models together; an output from a flopped output in verilated model 1 will change immediately after the posedge of the clock and be seen on the negedge's eval of verilated model 2 input, then the next posedge of model 2 will properly see the state.

smcpeak, 2020-02-26:
Hmmm.  In my real circuit, eval() is being called in the way you suggest.  The simulation loop comes from:

https://github.com/ucb-bar/riscv-sodor/blob/master/emulator/common/emulator.cpp

The reason there is still an issue, I think, is that the circuit uses the DPI-C interface for one of its components (called SimDTM).  In my reduced example, for simplicity, I'm driving the circuit inputs from the main simulation loop, but in Sodor the inputs in question come from SimDTM.  It will be more complicated (it will use DPI-C), but I will look into creating a new reproducer that conforms to the eval requirements you've stated.

wsnyder, 2020-02-26:
A DPI-C routine called inside an always_comb (or equiv) should get scheduled correctly, this is different code than the input propagation graph, so could be something else.  Thanks for working at isolating this.

smcpeak, 2020-02-29:
Ok, I have a new example: [cse-example2.tar.gz](https://github.com/verilator/verilator/files/4270910/cse-example2.tar.gz)

When run, this prints:
```
Top.v: state=0 outercond=1 innercond=0
Top.v: state=1 outercond=0 innercond=1
```

This seems wrong because DebugModule.v calculates the two conditions like this (same as above):
```
  assign op2_and_addr17 = op2  &  addr17;
  assign outercond      = op2_and_addr17 & valid;
  assign innercond      =         addr17 & valid;
```

and then exports them as output:
```
  assign out_outercond  = outercond;
  assign out_innercond  = innercond;
```

Top.v then prints them:
```
    $display("Top.v: state=%d outercond=%d innercond=%d",
      state, outercond, innercond);
```

It should never be the case that `outercond` is true while `innercond` is false since `outercond` is a logically stronger condition.

What is different from the previous example is the inputs are driven by SimDTM, which uses the DPI-C interface.

sim.cpp contains the top-level driver, which I think now clocks the circuit the way you indicated it should:
```
  VTop *top = new VTop;

  // Prepare for first work cycle.
  top->clock = 0;
  top->eval();

  // Bug appears here: outercond=1 innercond=0
  top->clock = 1;
  top->eval();

  // Prepare for another cycle.
  top->clock = 0;
  top->eval();

  // Run it.
  top->clock = 1;
  top->eval();
```

wsnyder, 2020-03-01:
Had to look at the debug dumps to figure this out as certainly not obvious!

Your code has a race.  This block:

    always @(posedge clk) begin
       debug_tick(_debug_req_valid,
                 _debug_req_bits_addr,
                 _debug_req_bits_op
                 );
    end

Is not using delayed assignments (of course) so may produce/consume signals racing with the consuming posedge block.

Note you had this for other simulators:

      assign #0.1 debug_req_valid     = _debug_req_valid;
      assign #0.1 debug_req_bits_addr = _debug_req_bits_addr[6:0];
      assign #0.1 debug_req_bits_op   = _debug_req_bits_op[1:0];

This was presumably to fix the races.  As Verilator ignores delays, this doesn't help Verilator with the races.  If you use this it works ok:

    always @(negedge clk) begin
      debug_req_valid     <= _debug_req_valid;
      debug_req_bits_addr <= _debug_req_bits_addr[6:0];
      debug_req_bits_op   <= _debug_req_bits_op[1:0];

smcpeak, 2020-03-01:
Thank you for the explanation!  I confirm that replacing the three time-delayed assignments with non-blocking assignments on the negative clock edge resolves the problem in the second example.  My apologies for using a buggy example; I'm new to Verilog.  (I didn't even know until now what those time delays were syntactically; this code was originally written by someone else.)

Might it be possible for Verilator to report the race condition in a warning?  (I see it already reports the unsupported time delays with `-Wall`.  I'm wondering if a more direct detection is possible.)

Regarding the first example, you mentioned adding documentation about proper usage of `eval()`, which seems like a good idea.  I'll note that in one of the examples from the manual:

https://www.veripool.org/projects/verilator/wiki/Manual-verilator#CONNECTING-TO-C

the harness code raises both `reset_l` and `clk` at the same time when `main_time` is 11, which seems to be against the rules as you've explained them above:

```
            while (!Verilated::gotFinish()) {
                if (main_time > 10) {
                    top->reset_l = 1;   // Deassert reset
                }
                if ((main_time % 10) == 1) {
                    top->clk = 1;       // Toggle clock
                }
                if ((main_time % 10) == 6) {
                    top->clk = 0;
                }
                top->eval();            // Evaluate model
                cout << top->out << endl;       // Read a output
                main_time++;            // Time passes...
            }
```

In addition to fixing the example, that place in the manual might be a good place to explain the `eval` and clock rules.

commit c108f5de (2020-03-06): Commentary, Closes #2177.
closed 2020-03-06 by wsnyder