verilator/verilator #2178: Error in cmd make -j -f Vour.mk Vour__ALL.a

Sangs19, 2020-02-25:
[image]
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this yourself?

wsnyder, 2020-02-25:
Looks like your $SYSTEMC variable isn't pointing at the SYSTEMC kit but rather an include file.  Check all your environment variables point to the right thing.

BTW for the future when possible please insert text rather than an image, as text allows cut & paste and the search engines to properly index it.

closed 2020-02-25 by wsnyder

Sangs19, 2020-02-25:
> [image]

Thanks. Cleared above error but the next command is not giving output from the manual.

make -j -f Vour.mk ../sc_main.o verilated.o 

is giving a statement of 

make: no rule to make target '../sc_main.o'

wsnyder, 2020-02-25:
You can't put a path on make targets, just use sc_main.o