verilator/verilator #2179: Unable to execute this g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o -o Vour -lsystemc

Sangs19, 2020-02-25:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this yourself?

make -j -f Vour.mk ../sc_main.o verilated.o

is giving a statement of

make: no rule to make target '../sc_main.o'

Also with g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o
-o Vour -lsystemc getting error ../sc_main.o no such file

wsnyder, 2020-02-26:
Probably because you don't have a sc_main.cpp or at least not one in that directory?

closed 2020-02-26 by wsnyder

Sangs19, 2020-02-26:
> Probably because you don't have a sc_main.cpp or at least not one in that directory?

Is it sc_main.cpp or sc_main.o in above cmd? Also how to generate sc_main.o file?

I am in learning stage and following the verilator manual

wsnyder, 2020-02-26:
The manual creates it, perhaps you missed that step.

Sangs19, 2020-02-26:
> make -j -f Vour.mk ../sc_main.o verilated.o
> 
> is giving a statement of
> 
> make: no rule to make target '../sc_main.o'
> 
> Also with g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o
> -o Vour -lsystemc getting error ../sc_main.o no such file

Thanks for solving the query. Now I am getting error as -lsystemc cannot be found