verilator/verilator #2180: Problem in executing g++ -L$SYSTEMC_LIBDIR ../sc_main.o Vour__ALL*.o verilated.o -o Vour -lsystemc

Sangs19, 2020-02-28:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

May we assist you in trying to fix this yourself?
Above statement is giving error of getting error of "cannot find lsystemc" when executed without the path ../ in sc_main.o

Else if executed exactly the command then getting error of ../sc_main.o no such file or directory.

Please help me out

wsnyder, 2020-02-28:
As you said earlier you are new at this, you may find this helpful in learning how to more quickly get to the answers and debug https://blog.hartleybrody.com/debugging-code-beginner/

Walking through this as an example:

Apply their 4: The message is saying something about systemc isn't found.

Apply their 5: google the message: See third hit
https://stackoverflow.com/questions/16710047/usr-bin-ld-cannot-find-lnameofthelibrary

So the message says that GCC isn't finding the library.

So, perhaps systemc isn't installed, or isn't installed where you are pointing to it with SYSTEMC_LIBDIR, and/or perhaps you skiped some install steps?

If that doesn't help you find it, when filing a issue (or replying to this one), apply their 10, and include what are 2-3 things you tried to fix it that didn't work?

wsnyder, 2020-02-28:
P.S. note "make -C" changes directories, so you need to do that if executing commands directly.

wsnyder, 2020-03-02:
Assuming fixed it.

closed 2020-03-02 by wsnyder

Sangs19, 2020-03-13:
> As you said earlier you are new at this, you may find this helpful in learning how to more quickly get to the answers and debug https://blog.hartleybrody.com/debugging-code-beginner/
>
> Walking [...]

I am still stuck at the same command. I tried using linking command
sudo ln -s /usr/lib/lsystemc.so.1.2.8 /usr/lib/libsystemc.so 
but could not solve it

wsnyder, 2020-03-13:
Most likely your SystemC is incorrect for your system. Either 1. where you got it from was broken or inappropriate for your system (& ask them), or 2. if you built it yourself you perhaps built on a different system than you are running on now, or had a very different compiler due to e.g. the environment or build user being different - don't do that ;)

Sangs19, 2020-03-15:
> As you said earlier you are new at this, you may find this helpful in learning how to more quickly get to the answers and debug https://blog.hartleybrody.com/debugging-code-beginner/
>
> Walking [...]

I am still stuck at the same command. I tried using linking command
sudo ln -s /usr/lib/lsystemc.so.1.2.8 /usr/lib/libsystemc.so 
but could not solve it

> Most likely your SystemC is incorrect for your system. Either 1. where you got it from was broken or inappropriate for your system (& ask them), or 2. if you built it yourself you perhaps built on a [...]

Thanks for the support. Finally got it.. The problem was with systemc in my system.